ckb-next/ckb-next #276: Can't find the animation directory after latest update.

bicarbonate, 2018-10-16:
I have not tried to build from source.. I'm unsure of what will happen since I'm using the copr build. And mentioning that.. I'm not sure where to correct post this issue..

### Source

copr build on F28

## What is the program's current behavior?

The <insert name> script could not be loaded. Please check your animation directory.

## What is the expected behavior of the program?

Loads animations correctly.

## Any other notes & comments?

---

# Something else

tatokis, 2018-10-16:
Try running `ls /usr/libexec/ckb-next-animations/` and paste the output

BumblingCoder, 2018-10-16:
I see the same issue. On my machine running that ls command gives
`gradient  heat  mviz  pinwheel  rain  random  ripple  wave`

In the application settings dialog it says that it is looking for the animations scripts in `/usr//usr/libexec/ckb-next-animations`, which seems to be the problem.

tatokis, 2018-10-16:
@hevanaa Can you please look into this?

bicarbonate, 2018-10-16:
Awesome, thank you. Updated and now I'm Rainbow again.

bicarbonate, 2018-10-16:
Issue resolved.

bicarbonate, 2018-10-16:
Crap sorry.

closed 2018-10-16 by bicarbonate

ssrebelious, 2019-03-11:
Having the same issue here after update to 0.4.0. I'm using [rpm for OpenSUSE Leap 15](https://build.opensuse.org/package/show/hardware/ckb-next). Is there a way to set correct path manually?

hevanaa, 2019-03-11:
You could notify them that their spec needs to be corrected. It should have "-DCMAKE_INSTALL_LIBEXECDIR=libexec", not "-DCMAKE_INSTALL_LIBEXECDIR=%{_libexecdir}"

Espionage724, 2019-05-17:
> Having the same issue here after update to 0.4.0. I'm using [rpm for OpenSUSE Leap 15](https://build.opensuse.org/package/show/hardware/ckb-next). Is there a way to set correct path manually?

I submitted a bug report here about it: https://bugzilla.opensuse.org/show_bug.cgi?id=1135528

bmwiedemann, 2019-05-29:
> You could notify them that their spec needs to be corrected. It should have "-DCMAKE_INSTALL_LIBEXECDIR=libexec", not "-DCMAKE_INSTALL_LIBEXECDIR=%{_libexecdir}"

This is not how it is supposed to work, because on openSUSE, it would have to be DCMAKE_INSTALL_LIBEXECDIR=lib
that is why we have these macros to abstract away the distribution differences.