ckb-next/ckb-next #278: Driver inconsistently recognizing keyboard on macOS Mojave

washedgram, 2018-10-23:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2503950/ckb-next-dev-detect-report.gz)

```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

Program very inconsistently recognizes keyboard on two different computers, both running macOS mojave 10.14. Sometimes the keyboard won't even be recognized by the computer itself after having installed ckb-next 

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

washedgram, 2018-10-23:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2503960/ckb-next-dev-detect-report.gz)

not sure if this got uploaded the first time

tatokis, 2018-10-23:
Try updating to 0.3.2 and see if that changes anything.

washedgram, 2018-10-23:
How exactly do I update again?

tatokis, 2018-10-23:
https://github.com/ckb-next/ckb-next/releases/download/v0.3.2/ckb-next_v0.3.2.dmg
Install that, and maybe reboot afterwards.

washedgram, 2018-10-23:
Fixed it. Thanks!

closed 2018-10-23 by tatokis

washedgram, 2018-10-31:
Having the same issue again. I think it’s happening after I unplug from Mac and use on windows then try to use on Mac again. Should I submit a support ticket again?

tatokis, 2018-11-01:
Please generate another report and attach it to this issue