ckb-next/ckb-next #279: Keep on getting warning like that and with no device connected all the time

StevenGAO95, 2018-10-23:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

[image]
I just can not find the way to admit the system extension to solve this warning.

StevenGAO95, 2018-10-23:
My keyboard is K70 RGB RapidFire.

tatokis, 2018-10-23:
First, unplug and replug your keyboard.
Then you should be able to find the allow button under Security & Privacy:
[image]

StevenGAO95, 2018-10-23:
[image]
I tried to unplug and replug it and the warning disappeared but it still can not be detected. It is unable for me to input with the light on.

tatokis, 2018-10-23:
Click the generate report button and upload the report to this issue.

StevenGAO95, 2018-10-23:
[ckb-next-report-1540321607.gz](https://github.com/ckb-next/ckb-next/files/2507623/ckb-next-report-1540321607.gz)
Here is the report. Thank U so much.

tatokis, 2018-10-23:
If your keyboard was plugged in when you generated that report, then it appears that macOS is not recognising it at all, so ckb-next can't do much about that.

Try plugging both USB cables to your macbook and see if that helps.

Do you know if the keyboard work on windows with the corsair software?

StevenGAO95, 2018-10-23:
The cable for the extending usb port works well. I will try on windows and show the result later.

StevenGAO95, 2018-10-23:
I tried on another Macbook Pro and it does not work either. But I have used this keyboard during last month, so I don't know what happened yesterday when I unplug it and replug it, it does not work anymore.

tatokis, 2018-10-23:
Unplug the keyboard, hold down the Esc key on it, and with the Esc key held down, plug the keyboard in to your computer.
Wait about 3 seconds and release the Esc key. See if that fixes it.

StevenGAO95, 2018-10-23:
It works. Thank you so much!

tatokis, 2018-10-24:
Good to hear. Be sure to upgrade to the latest firmware if you haven't already.

closed 2018-10-24 by tatokis