ckb-next/ckb-next #281: Quickinstall host configuration issue

timmyolson, 2018-10-24:
# Bug report

## General information

Devices:
- Corsair Gaming K95 RGB PLATINUM Keyboard
- M65 Pro RGB Mouse

### Source
  * built manually using quickinstall script

## Logs & crash reports

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2512390/ckb-next-dev-detect-report.gz)

I apologize for all the noise in here I was really frustrated and just keep trying to get it to work the other day by brute forcing it.

Another note about the journal from the current boot, despite never working on this system when I woke the machine from hibernation the mouse and keyboard loaded, accepted the default lighting patterns and worked for about 1 minute. I can't decipher if the partial success is in the journal.

## What is the program's current behavior?

Starting the service unit will crash loop with:
```
Oct 24 16:08:25 baldr ckb-next-daemon[1080]: [W] os_usbrecv (via firmware.c:24): Interrupt cond error 110
Oct 24 16:08:25 baldr ckb-next-daemon[1080]: [W] _start_dev (device.c:43): Unable to load firmware version/poll rate
Oct 24 16:08:25 baldr ckb-next-daemon[1080]: [W] os_usbrecv (via firmware.c:24): Interrupt cond error 110
Oct 24 16:08:25 baldr ckb-next-daemon[1080]: [W] _start_dev (device.c:43): Unable to load firmware version/poll rate
```

## What is the expected behavior of the program?

## Any other notes & comments?

My work board K70, once had `Unable to load firmware version/poll rate` it was set in BIOS mode, I'm confident this isn't the issue because I can use ckb-next without any flaws on a different computer.

That other computer is a Lenovo laptop that also runs Ubuntu 18.04. Besides being different hardware the only real differences in OS configuration might be that that one started its life as Ubuntu 17.04 and was upgraded over the years.

Might be more info than you care about but here it is working on that system:
```
Oct 21 18:15:39 orthrus systemd[1]: Started Corsair Keyboards and Mice Daemon.
Oct 21 18:15:39 orthrus ckb-next-daemon[1929]:     ckb-next: Corsair RGB driver 0.3.2-4-g84ffa81
Oct 21 18:15:39 orthrus ckb-next-daemon[1929]: [I] Root controller ready at /dev/input/ckb0
Oct 21 18:15:39 orthrus ckb-next-daemon[1929]: [I] Connecting Corsair Gaming K95 RGB PLATINUM Keyboard at /dev/input/ckb1
Oct 21 18:15:39 orthrus ckb-next-daemon[1929]: [I] Connecting Corsair Gaming M65 Pro RGB Mouse at /dev/input/ckb2
Oct 21 18:15:39 orthrus ckb-next-daemon[1929]: [I] Starting input thread for /dev/input/ckb2
Oct 21 18:15:39 orthrus ckb-next-daemon[1929]: [I] Starting input thread for /dev/input/ckb1
Oct 21 18:15:40 orthrus ckb-next-daemon[1929]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 06 02 01 01 1c 1b 2e 1b 01 01 01 30 c1 ff 40 01 56 00 01 01 00 00 00 00 00 00 00 00 01 00 00 01 00 00 02 01 00 01 aa 00 6c dc 02 00 00 00 00 00 00 00 00 00 00 00 00 00
Oct 21 18:15:40 orthrus ckb-next-daemon[1929]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 07 01 04 00 1c 1b 2d 1b 01 01 00 00 c0 ff 40 00 03 00 00 00 02 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Oct 21 18:15:40 orthrus ckb-next-daemon[1929]: [I] Setup finished for /dev/input/ckb1
Oct 21 18:15:42 orthrus ckb-next-daemon[1929]: [I] Setup finished for /dev/input/ckb2
```

Finally I can bring my K70 home if it's needed for debug, I would wonder if that one in addition to this keyboard and mouse would also fail. (I can do that in the next 24 hours)

---

Finally I apologize because given the fact that this keyboard works flawlessly on two different computers, my work Mac, Ubuntu laptop, etc it's most just some system config issue, and a waste of time.

ZirconiumX, 2018-10-24:
>interrupt cond error 110

In English, this means we were expecting the keyboard and mouse to reply, but they didn't. The firmware versions you have are fairly well supported, which makes this very unusual.

The identification packets given on your other laptop look okay to me, suggesting it's an issue with your computer.

As a side note, ckb can detect and release a keyboard in BIOS mode.

timmyolson, 2018-10-25:
The only thing I had not tested and verified on this machine was the motherboard. I had no issues running ckb on it previously.

Just to be sure I flashed it and reinstalled all the drivers. It appears to work now. USB driver must have gotten out of wack.

All my old macros, lighting, etc work. This is great, now I feel like I have all my digits again. Thanks for your insight.  :+1:

closed 2018-10-25 by timmyolson