ckb-next/ckb-next #286: keyboard not recognized after use w/PC

washedgram, 2018-11-01:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2536843/ckb-next-dev-detect-report.gz)

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

Keyboard (corsair RGB strafe) inconsistently recognized by Mac (macOS mojave), never recognizes CKB program after switching use of the keyboard from a PC.  

### Source

built manually

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

tatokis, 2018-11-01:
Do you boot windows on your mac, or do you physically unplug and replug the keyboard to another computer?
I've been seeing this issue for a while but I have been unable to reproduce it in order to try to fix it.

washedgram, 2018-11-01:
I boot windows on Mac ( I have 2 Macs that I switch the keyboard between) and after using it on windows boot and re-plugging into another machine (no partition) is when it’s first unresponsive. It takes a subsequent reboot of the Mac with the windows partition for the program to recognize the keyboard, then it works fine until I use it in the windows boot again.

tatokis, 2018-11-02:
So, looking at your logs, I see
```
[W] add_hid (usb_mac.c:848): Tried to set up ifhid[1] for device ckb1, but it was already set up. Skipping...
```
Which partially explains why it doesn't work.

https://github.com/ckb-next/ckb-next/blob/6c83cdf284b2e07c20dac733b91492a7c5598d3e/src/daemon/usb_mac.c#L851-L855
I think the comment sums up my thoughts.

Anyway, can you install this build, reproduce the issue, and then generate another report while the issue is happening?
http://ckb-next.info/mac/ckb-next_v0.3.2-8-g6c83cdf_usbdbg.dmg

```
e3516493b9c0721df9f126598f8c681d866eb94f842c778addd8abf1f6409baf  ckb-next_v0.3.2-8-g6c83cdf_usbdbg.dmg
```

washedgram, 2018-11-02:
I uninstalled the old version and installed this build, now the app won’t open at all.

tatokis, 2018-11-02:
Apologies.
Try this one http://ckb-next.info/mac/ckb-next_v0.3.2-11-g0429e1a_usbdbg.dmg

```
7afdf72b1e260109aff6e9e1e0251f0e48a41b1df7ee71e66ab2e49b390731b4  ckb-next_v0.3.2-11-g0429e1a_usbdbg.dmg
```

washedgram, 2018-11-04:
Seems to be working fine now. Will update if it doesn’t 

Grahame

alexclifford, 2018-11-28:
I too am running v0.3.2 and see this problem when physically unplugging my K70 RGB from a Windows PC into an Apple Macbook Pro. Using the above DMG allows me to restart ckb-next and it works instantly (Looks to be https://github.com/ckb-next/ckb-next/tree/mac-avtopcase that has the fixes?)

ETA on this being released in a version?

tatokis, 2018-11-29:
@alexclifford that branch is unrelated, and the build you installed does not have that code.

Additionally, this makes absolutely no sense whatsoever.
I never made a fix for this issue, because I never figured out what causes it.

The only two daemon-related changes between 0.3.2 and the commit the above build was based on are unrelated to the usb code.
On top of that, only thing I did extra was add a few lines that print the state of the handles.

Perhaps it's a timing issue? I have absolutely no idea.

I've been looking for the changes I did to the code, but I think I have discarded them during one of my `git reset --hard` shenanigans.

TLable, 2019-04-20:
My software is working and tells me there is no device connected.  

I haven't tried rebooting to find if my drivers recognize device yet. 

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/3100509/ckb-next-dev-detect-report.gz)

tatokis, 2019-04-20:
@TLable you should open a new issue. It looks like OpenCorsairLink is having trouble talking to one of your other devices, so you should try disabling it for now and see if it helps with the issue.

Also, closing this issue because 0.4.0 has been released.

closed 2019-04-20 by tatokis