ckb-next/ckb-next #287: Cannot uninstall properly

danilodequeiroz, 2018-11-04:
The uninstall procedures only disabled/ removed the daemon, but the cbk-next gui is still available on overview search on ubuntu 18.04.

AaronTakizad, 2018-11-04:
I resolved this issue by removing `ckb-next-*` executables from `/usr/local/bin` (from memory?). I also found residual animations data in `/usr/libexec/ckb-animations`. Make sure to also remove the program from your startup applications, as this is also missed by the uninstaller.

tatokis, 2018-11-26:
Assuming you mean just the desktop entry, I just tested it and the .desktop file was removed for me.
Are you sure you just didn't need to reboot/wait for the cache to get cleared?
Can you reproduce this?

closed 2022-10-03 by tatokis