ckb-next/ckb-next #288: Uinput module could not be loaded after reboot

kraxarn, 2018-11-04:
After installing the software (0.3.2), it worked fine after several reboots, but after rebooting my computer today, the driver refuses to load. As far as I know, the system hasn't updated or anything between the reboot. When starting the software, I only get the error `Warning: The uinput module could not be loaded.`. I tried rebooting and reinstalling everything, but the issue is still there. I also tried installing the latest version from git, but then the daemon fails to load instead. The only error I got while reinstalling 0.3.2 was:
```
-- systemd detected
-- Generating and importing ckb-next-daemon.service (systemd)
[sudo] password for root: 
inactive
Failed to get unit file state for ckb-daemon.service: No such file or directory
inactive
Failed to get unit file state for ckb-next-daemon.service: No such file or directory
```
It's installed with compiling the source code as I couldn't find a binary for openSUSE Tumbleweed.

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2545841/ckb-next-dev-detect-report.gz)

kraxarn, 2018-11-04:
I'm pretty sure the kernel hasn't been modified. I don't have a rollback option and DKMS didn't kick in after starting it today.
It's a newly installed computer, so it was fully updated before I installed ckb and I haven't done any major updates on it. To try things out though, I tried it on another computer, running the same OS and kernel, and it works fine (though, it's using ckb-next 0.3.0).

ZirconiumX, 2018-11-04:
What does `modprobe uinput` (as root) say?

kraxarn, 2018-11-04:
It doesn't output anything and the output of `modprobe uinput --first-time` is `modprobe: ERROR: could not insert 'uinput': Module already in kernel`.

tatokis, 2018-11-04:
Well, we don't use --first-time, so can you try `modprobe uinput` and then immediately after `echo $?` ?

kraxarn, 2018-11-04:
It outputs 0.

I tried a couple of things and it randomly started working, but after a reboot it stopped working again.
I noticed that, most of the time, the mouse and keyboard are frozen after startup and need to be re-plugged in to work again. When they freeze, ckb-next doesn't work. I remember having a similar issue a while back with the old ckb, but can't remember how I fixed it.

kraxarn, 2018-11-18:
I think I managed to find a fix.
According to some random Corsair forum post, if you have USB 3, it's recommended to just use one of the USB cables from the K95 in a single USB 3 port. This fixes some issues on Windows, but was most likely causing this issue on Linux. I tried connecting both USB cables from the K95 (to USB 2) and haven't had the issue since. I'll close the issue for now, but if it happens again, I'll reopen it.

The error message is still misleading though.

closed 2018-11-18 by kraxarn