ckb-next/ckb-next #289: daemon crashes immediately in Fedora 29

jwright1985, 2018-11-06:
# Bug report

## General information
daemon crashes on launch in Fedora 29 kernel 4.18.16-300.fc29.x86_64

### Source
- pulled November 4 about 5PM eastern, installed using quickinstall

## Logs & crash reports
]$ git clone https://github.com/ckb-next/ckb-next.git
]$ cd ckb-next/
]$ ./quickinstall
<prompts for sudoer password>

- quickinstall failed to create the systemd  unit file, so I created it and replaced the ExecStart line from the ckb-next-daemon.service.in with ExecStart=/usr/local/bin/ckb-next-daemon and placed the unit file in /etc/systemd/system
- started the daemon
- daemon segfaults and dumps a core
]# abrt-cli list
id 5e0c959f79e3bbd3ddc561265f2e05829242dba5
reason:         _start_dev(): ckb-next-daemon killed by SIGSEGV
time:           Tue 06 Nov 2018 05:01:16 PM EST
cmdline:        /usr/local/bin/ckb-next-daemon
uid:            0 (root)
count:          35
Directory:      /var/spool/abrt/ccpp-2018-11-06-17:01:16.275191-4626

##Uploads
[ccpp-2018-11-06-17:01:16.275191-4626.zip](https://github.com/ckb-next/ckb-next/files/2555369/ccpp-2018-11-06-17.01.16.275191-4626.zip)
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2555370/ckb-next-dev-detect-report.gz)

## What is the program's current behavior?
- segfaults almost immediately and devices stop responding entirely (no keyboard input, no mouse movement)
- does not apply settings to devices

## What is the expected behavior of the program?
- runs normally

## Any other notes & comments?
- keyboard and mouse have to be physically disconnected and reconnected to regain basic functionality after the crash
- All device firmware is up to date. Booted a windows box and updated firmware on keyboard and mouse to latest to ensure.
- Issue persists through reboots and attempts to restart the daemon

ZirconiumX, 2018-11-07:
You're going to have to join us on IRC; there's a blue Freenode button that will get you to webchat. Make sure you mention "ZirconiumX" so that I don't miss you.

tatokis, 2018-12-09:
Fixed in de483fe7a414

closed 2018-12-09 by tatokis