ckb-next/ckb-next #294: Disable Mouse Acceleration Gone

JonathanZelaya, 2018-11-12:
# Feature request

## What is the program's current behavior?
The "Disable Mouse Acceleration" toggle is missing in 0.3.2. It was present in 0.3.0.
## What would you like it to do instead?
Have the option to disable acceleration again.

tatokis, 2018-11-12:
Unfortunately since macOS Sierra and higher, our old code that disabled mouse acceleration was broken by Apple, and since Mojave it stopped functioning completely.

The new way we send input events to the operating system does not allow us to disable mouse acceleration, so you will need to disable it the same way you would with any other mouse.

```
defaults write .GlobalPreferences com.apple.mouse.scaling -1
```
Something like that in a terminal should disable it.

closed 2018-11-12 by tatokis

JonathanZelaya, 2018-11-12:
Thanks for letting me know.