ckb-next/ckb-next #295: Target Display Mode Key Sequence Support (Command-F2)

brianhall12345, 2018-11-21:
I'm running ckb-next 0.3.2 on Mac OS Mojave on a iMac (27 inch, Late 2012). I use target display mode to use the large 27 inch display from my iMac for my MacBook Pro. I recently switched to using a Corsair K65 keyboard and Corsair Dark Core mouse connected to a switch that allows me to share the keyboard and mouse with the laptop and desktop.

The issue I have is Target Display Mode seems to be a special key sequence that is only transmitted by actual Apple keyboards.  The key combination is Command F2, but it doesn't work with any keyboard other than an apple keyboard.  So at present, I still have my wireless apple keyboard sitting to the side, and its sole purpose is to toggle target display mode.

Is it possible we can uncover what signal is being sent via that key combination, and replicate it in ckb-next?

ZirconiumX, 2018-11-21:
I suspect it's a virtual key press that we currently don't have. Not that Apple are much help with this kind of thing. @tatokis - good luck.

tatokis, 2018-11-29:
Looks like it's closely related to #284 .

@brianhall12345 What happens if you install http://ckb-next.info/mac/ckb-next_v0.3.2-12-g0a9ca07_avtopcase.dmg , bind F2 to brightness up in ckb-next and then try again?

brianhall12345, 2018-11-30:
Just gave it a try, it doesn't seem to have an impact.

closed 2022-03-05 by tatokis