ckb-next/ckb-next #296: [Bug] K66 not detected by ckb-next

TehLlama, 2018-11-22:
# Bug report

## General information
I'm trying to get my Corsair K66 working with Linux (Ubuntu 18), and the driver won't consistently detect the keyboard - but Ubuntu will with the ```lsusb``` command
### Source
Built program manually with the git source and ```quickinstall``` script, version 0.3.2

## Logs & crash reports
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2609050/ckb-next-dev-detect-report.gz)
Here is the terminal for running ```lsusb```:
```
$ lsusb
Bus 002 Device 002: ID 8087:8001 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8009 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 003: ID 1038:1250 SteelSeries ApS 
Bus 003 Device 002: ID 046d:c069 Logitech, Inc. M-U0007 [Corded Mouse M500]
Bus 003 Device 007: ID 1b1c:1b41 Corsair 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Here is proof that the daemon is running:
```
ckb-next/build/bin$ sudo ./ckb-next-daemon 
    ckb-next: Corsair RGB driver 0.3.2-13-g448f337
[F] ckb-next-daemon is already running (PID 770). Try `killall ckb-next-daemon`.
[F] (If you're certain the process is dead, delete /dev/input/ckb0/pid and try again)
ckb-next/build/bin$ ps 770
  PID TTY      STAT   TIME COMMAND
  770 ?        Ss     0:00 /usr/local/bin/ckb-next-daemon

```
## What is the program's current behavior?
ckb-next loads normally when opened, and the daemon seems to run on startup.

VERY RARELY, ckb-next will actually detect the keyboard after either a restart or some random occasion; when this happens, the keyboard will work without issues until the next restart. If I can get this to happen again, I could post more information, but I need to know what to share.
## What is the expected behavior of the program?
Self-explanatory; it should recognize the keyboard (K66) all the time.
## Any other notes & comments?
So I'm willing to help test making the K66 work with ckb-next, but I need to know if this isn't just an issue with my setup and usage of ckb-next first. I haven't seen much discussion of this keyboard, so I'm not sure if there's been much testing done.

Also, if you need my help testing, the testing environment will have to be different than my current environment. But, the only change would be from Ubuntu 18 to Ubuntu 16.04.

tatokis, 2018-11-22:
This is actually corsair's firmware not being too compliant, requiring specific quirks set in the linux kernel to be detected properly.
You can try plugging the keyboard to a different usb controller, different usb ports in general, or a powered usb hub.

If none of that helps, you will need to upgrade to the latest linux kernel (preferably from ukuu) and then adding this quirk https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbcore-quirks

If that quirk does fix it, do report back so that we can then hopefully try to get the quirk submitted to the linux kernel so that the external quirk isn't required.

As a sidenote, @ZirconiumX do you think it may be worth adding a warning that quirks may be required when we get `Unable to read endpoint count from udev, assuming 0` ?

```
Nov 22 13:18:15.320067 jacob-build-pc ckb-next-daemon[3130]: [I] Connecting Corsair Gaming K66 Keyboard at /dev/input/ckb1
Nov 22 13:18:15.320067 jacob-build-pc ckb-next-daemon[3130]: [E] os_setupusb (usb_linux.c:597): Unable to read endpoint count from udev, assuming 0
Nov 22 13:18:15.320067 jacob-build-pc ckb-next-daemon[3130]: [I] Attempting reset...
Nov 22 13:18:15.796064 jacob-build-pc ckb-next-daemon[3130]: [I] Reset success
```

TehLlama, 2018-11-23:
I'm starting to work on updating the kernel and adding this to the ```cmdline```:
```
usbcore.quirks=1b1c:1b41:g
```
I will update this issue later with the results.

TehLlama, 2018-11-23:
Attempted to run Linux Kernel v4.19.3. Linux itself was slow and sluggish, and the updated kernel and ```cmdline``` didn't fix the issue. I'm gonna try some other versions of the Linux kernel to see what I can do.

Let me also double check to see if the line I added to ```cmdline``` is correct (should be no problem with it).

tatokis, 2018-11-23:
Generate another report please

TehLlama, 2018-11-23:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2609702/ckb-next-dev-detect-report.gz)

I restarted on the latest kernel (4.19.3) and my K66 started working! This is the report from that boot.

I will restart a few times to see if it keeps working on boot, and I'll also add the following to ```cmdline``` to try and make the kernel load and perform better:
```
usbhid.quirks=0x1b1c:0x1b41:0x20000408
```

TehLlama, 2018-11-23:
So the keyboard works after adding the quirk and updating the kernel. Thanks for the help!

However, Linux is really sluggish now, and I'm not sure if its the kernel version or something else about the quirk I had to add. Do you have any suggestions for fixing this?

tatokis, 2018-11-23:
I assume you use nvidia graphics. The version of the graphics driver ubuntu installs probably doesn't support the 4.19 kernel.
If that is the case, you can add the "graphics ppa" and install the driver from it.

If your graphics card is supported by the latest drivers you should be able to just do
```
sudo add-apt-repository ppa:graphics-drivers/ppa
sudo apt update
sudo apt install nvidia-415
```
and then reboot.

The PPA is here https://launchpad.net/~graphics-drivers/+archive/ubuntu/ppa but the commands above add it automatically.

TehLlama, 2018-11-23:
You were right about just updating the graphics driver to one that worked with the kernel. The method you posted didn't work for me for some reason, but I managed to get the graphics driver installed myself. 

Thanks for all the help! Feel free to DM me if you need help testing the K66 keyboard or anything like that.

closed 2018-11-23 by TehLlama

ZirconiumX, 2018-11-24:
@TehLlama You mentioned that 4.19.3 was sluggish. You might be [bitten by STIBP](https://www.phoronix.com/scan.php?page=article&item=linux-420-stibp&num=1).