ckb-next/ckb-next #297: Support for Portuguese ISO Layout

Madril, 2018-11-23:
Is there a way for ckb-next to support the Portuguese (pt-PT) ISO layout?

tatokis, 2018-11-26:
Is this on mac?
All we do is pass through keystrokes, and the layouts are purely visual, so it's up to the OS to handle it.

macOS doesn't seem to have support for the layout, so I don't think there's much we can do about it, and I'd rather not hack around the code with permanent key remaps.

Madril, 2018-11-26:
It is on a Mac.
Got it working kinda fine now. Only < and § are switched.

sidneywidmer, 2019-02-23:
@Madril I have the same problem with the Swiss layout. Did you find a fix for it? Everything works great but < and § are switched 🤔

sidneywidmer, 2019-02-23:
Just figured it out 🙈 on MacOS just force rerun the Keyboard assistant and everything works as expected 👍

closed 2022-03-05 by tatokis