ckb-next/ckb-next #298: CORSAIR K70 RGB MK.2 LOW PROFILE

charlie-cyw, 2018-11-23:
#### New device support request

CORSAIR K70 RGB MK.2 LOW PROFILE Mechanical Gaming Keyboard

Currently it shows No devices connected in ckb-next interface.

[ckb-next-report-1542995030.gz](https://github.com/ckb-next/ckb-next/files/2611440/ckb-next-report-1542995030.gz)

Many thanks 👍

tatokis, 2018-11-23:
I'll add it in a few minutes. Would you like a prebuilt package, or will you build the code yourself?

tatokis, 2018-11-23:
Either way, https://github.com/ckb-next/ckb-next/commit/864f91f81b80f0e6830b2631caaf27eab3cfb9ce

charlie-cyw, 2018-11-23:
Nice one, I'll build it and get back to you.

charlie-cyw, 2018-11-23:
It's solid!

tatokis, 2018-11-23:
If everything is working correctly, you can control all LEDs, and no keys are unresponsive, then please close this issue. As a sidenote, you can't save to hardware yet.

Thanks for testing

charlie-cyw, 2018-11-23:
Yeah, should be good so far. Thanks

closed 2018-11-23 by charlie-cyw