ckb-next/ckb-next #300: Strafe mk.2 how to change logo colors

rafaelpap, 2018-11-24:
# Feature request
It's possible to assign different colors to the logos?

## What is the program's current behavior?
Using the daemon I am unable to change the colors.
Using the gui only the right part (CORSAIR) changes color. The left one appears black.

## What would you like it to do instead?
To be able to change both parts.

## Any other notes & comments?
If I do

sed -i "s/k70mk2/strafe_mk2" /dev/input/ckb1/features

the gui allows to change the color of both parts.

ZirconiumX, 2018-11-24:
That shouldn't be hard to fix; probably @tatokis forgot to update the
keymap.

tatokis, 2018-11-24:
So... Which keyboard do you have?

rafaelpap, 2018-11-25:
I have the Corsair Strafe RGB MK.2 keyboard.

rafaelpap, 2018-11-25:
In keymap.cpp under the gui directory I find the lines
    case KeyMap::K70MK2:{
        // Same as K70 but buttons moved to the left, profswitch, and two logo lights (using m3, g11)
...
       // Add logo1 (m3)
        map["m3"] = {0, "Logo 1", "m3", 118, 0, NS, true, false};
        // Add logo2 (g11)
        map["g11"] = {0, "Logo 2", "g11", 138, 0, NS, true, false};

Maybe they should also go inside the 
   case KeyMap::STRAFE_MK2:{
?

tatokis, 2018-11-26:
That is exactly what this line does
https://github.com/ckb-next/ckb-next/blob/864f91f81b80f0e6830b2631caaf27eab3cfb9ce/src/gui/keymap.cpp#L571

I do, however, know what the issue is and I'll fix it.

commit 418af411 (2019-01-12): Fix STRAFE_MK2 logo

Closes #300