ckb-next/ckb-next #302: Change position of keys in the Strage RGB MK.2 keyboard

rafaelpap, 2018-11-26:
# Feature request
Change position of profshift,light & lock keys on Strafe RGB MK.2 keyboard
## What is the program's current behavior?
The gui shows the keys at the K70 MK.2 position. The Strafe has the keys starting on top of the F1 key.
## What would you like it to do instead?
I suggest to add the following lines in the keymap.cpp file
        // Move profswitch, light and lwin two block to the left
        map["profswitch"].x -=20;
        map["light"].x -= 20;
        map["lock"].x -= 20;

just before 
       map["lsidel"] = KStrafeKeys[0];
        map["rsidel"] = KStrafeKeys[1];

tatokis, 2018-11-26:
Do you want to open a pull request for me to merge, or do you just want me to just go ahead and change the code?

rafaelpap, 2018-11-26:
I must confess it's my second day at github. If you prefer I could try to open a pull request.

tatokis, 2018-11-29:
That's fine.
I mostly asked so that you get credited, but I did on my commit message.

In the future, pull requests would be preferred. Both so that you get credited, and that it's easier to review/maintain.

Thanks again!

commit 1851f03d (2018-11-29): Properly position Strafe MK2 Mute and Volup/dn

Thanks to @rafaelpap
Closes #302
closed 2018-11-29 by tatokis