ckb-next/ckb-next #303: CapsLock RGB indication changes before CapsLock is activated by the system

luccasclezar, 2018-11-28:
# Bug report

# Devices information

I uploaded the file even though I don't think it's needed, for this issue I think it suffices to say that I have a K70 plugged to a MacOS. But, here it is:

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2625788/ckb-next-dev-detect-report.gz)

## General information

### Source
Both pre-built and compiled from source (last commit is 0611711a2cb96d6a9e9b0f8b1b0429c9df3b5aa3) for MacOS.

## What is the program's current behavior?
The CapsLock RGB indication light changes its color with a quick tap on CapsLock key.

## What is the expected behavior of the program?
The CapsLock color doesn't change if the key is quickly tapped (as it is with a built-in MacBook keyboard), as MacOS has a time threshold to activate the CapsLock.

# Something else

When the CapsLock is quickly tapped, the system doesn't activate the CapsLock but the keyboard changes the key's color. This is very annoying as sometimes I accidentally press CapsLock but I can't know if the system activated it or not, resulting in the situation where I have to write something to know.

Sometimes this situation is quickly fixed because I was already writing, then just another quick tap on the CapsLock key solves it, but there are times when I accidentally press CapsLock and I have to explicitly go to a place where I can write only to check if it is toggled. This happens three to five times a day and it should be a pretty simple fix to only change the color when the key is held for like, 200ms, right?

I want to finish thanking everyone that maintains and updates ckb-next, to be able to use my K70 to its full potential on a MacOS is really nice. Thanks!

tatokis, 2018-11-29:
To sum up the situation, macOS doesn't have global caps lock between multiple keyboards, which means each keyboard has its own state.

In order to light up the caps lock key/led in ckb-next, we just toggle it on or off every time the key is pressed.
Now, as you have figured out, this tends to break easily.
The only way to read caps lock events was hacky, and we never implemented it.

It ended up breaking on Mojave anyway.

The good news is that VirtualHIDDevice now has a feature to compensate for this exact issue, which I assume can be implemented with relative ease in ckb-next.

From the 6.9.0 changelog: 

```
    LED has been supported on VirtualHIDKeyboard.
    A HID report {kHIDPage_LEDs, kHIDUsage_LED_CapsLock} will be posted to user space when the VirtualHIDKeyboard LED state is changed.
```
https://github.com/tekezo/Karabiner-VirtualHIDDevice/blob/master/NEWS.md#version-690

We currently use 6.8.0, so I'll definitely try to implement this as soon as possible.

luccasclezar, 2018-11-29:
Hi @tatokis, thank you for your reply.

But I think you misunderstood my issue. When I press CapsLock, it instantly lights up the key, but MacOS doesn't instantly turns on CapsLock in the system.

MacOS has a small delay to prevent accidental bumps on the key to turn on CapsLock. This means that, if the key is unpressed before the system's delay is done, the only thing that will change is the light. The system will not turn on CapsLock. This leaves the keyboard unsynchronized with the system, as the key light is on but the system's CapsLock is off.

tatokis, 2018-12-01:
While I did understand what you meant, I did go a bit off topic, but the underlying issue is basically the same.
I'll look to have a fix soon

TheManchineel, 2019-01-15:
Are there any updates/progress/ETA on this?

tatokis, 2019-01-16:
Not yet, apologies. I'm basically busy due to a combination of exams and working on about 4 more other ckb-related things.

I'll see if I can get this done by next Monday.

TheManchineel, 2019-01-16:
@tatokis No worries mate, actually thanks for making this superb piece of software!

Good luck with your exams by the way!

luccasclezar, 2019-01-25:
Thanks for the update @tatokis! And don't need to rush, if you still need time for your exams, feel free to delay this :)

tatokis, 2019-03-03:
Just wanted to mention I didn't forget about this. I've been preparing for the next release, and I am still working on this bug.

TheManchineel, 2019-03-04:
> 
> 
> Just wanted to mention I didn't forget about this. I've been preparing for the next release, and I am still working on this bug.

Thanks a lot for the dedication! Waiting for the release then.

tatokis, 2019-03-21:
I'm really struggling to get this to work. I can't get the hid code to receive any events.
Would help if Apple actually had proper documentation

luccasclezar, 2019-03-22:
Thank you again for the update and for your dedication @tatokis! 

I don't know a thing about this kind of programming so it's hard for me to help, but my understanding is that you're trying to receive the Caps Lock event fired from the system to change the lighting of the key. Wouldn't it possible to just figure out Caps Lock's activation delay and implement it manually on CKB? Like only activate Caps Lock when the key is pressed long enough using a timer or something like that instead of relying on a system event?

tatokis, 2019-04-18:
Unfortunately, I will end up resorting to adding a fixed delay and hope it works properly, as I can't figure out how to get this to work.

Again, I'm really sorry for these really long delays.

I'll try to get you a build to test in the upcoming days, as I am having trouble with my macOS installation.

luccasclezar, 2019-04-19:
No problem @tatokis, thank you very much for creating ckb and for your dedication!

I'll be waiting to test it out and tell you the result :smile:

closed 2022-03-05 by tatokis