ckb-next/ckb-next #310: 0.3.2 shows no devices connected after Mojave Update

gaoxing840, 2018-12-07:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2658779/ckb-next-dev-detect-report.gz)
Hi all, i have been using CKB for a long time. 
Was on the older version <0.2.9 and my RGB Strafe works perfectly with key binding and lighting animation.
Recently I have updated my Mac Pro Retina Early 2015 to Mojave.
I tried installing the latest 0.3.2 but the CKB doesn't recognise my keyboard and shows no devices connected.
I can't bind my keys, however i can use it normally.
I attached my report and hope you guys can help me figure it out.
Thank you!

ZirconiumX, 2018-12-07:
You need to re-enable the Karabiner kext; you can do this in System Preferences > Security.

florianorama, 2018-12-09:
Hello,

> You need to re-enable the Karabiner kext; you can do this in System Preferences > Security.

Do you mean to drag and drop the ckb-next app under Accessibility?

Thank you!

PS My first time installation…

beatlevic, 2018-12-11:
I'm having the same issue (no devices connected). I'm installing CKB for the first time (version 0.3.2) on Mojave 10.14.2.

tatokis, 2018-12-11:
You need to go to System Preferences > Security & Privacy and allow the extension by Fumihiko Takayama, by clicking the allow button on the bottom right.
If it doesn't appear, try unplugging and replugging your corsair devices.
You also need to either reboot or unplug and replug your devices after you've allowed it.
[image]

beatlevic, 2018-12-11:
Thank you for your quick response. In my case, I "Allowed" the developer already, but without success. I also tried rebooting, reinstalling and plugging my mouse (k65) in and out.

I'm on a macbook pro 13 inch 2018, with a usb-c to usb-a adapter for my k65 mouse.

I included my report here: 
[ckb-next-report-1544540641.gz](https://github.com/ckb-next/ckb-next/files/2668078/ckb-next-report-1544540641.gz)

tatokis, 2018-12-11:
@beatlevic Looking at the report, you have an M65 mouse. Only the M65 RGB and M65 RGB Pro are currently supported.

If I recall correctly, the M65 uses a different protocol, however we could try adding support for it if you want to spend time on it and help.

What features of it are you looking to configure?

tatokis, 2018-12-16:
You can run `echo "pollrate 2" > /var/run/ckb1/cmd` in the terminal with ckb-next running, and it will switch your mouse to 2ms pollrate. See #272

closed 2022-06-27 by tatokis