ckb-next/ckb-next #316: Add Corasir RGB Memory Support

TheBrokenRail, 2018-12-22:
# Feature request

## What is the program's current behavior?
It does not detect Corsair RGB RAM.

## What would you like it to do instead?
Detect and control RGB RAM.

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2704663/ckb-next-dev-detect-report.gz)

CalcProgrammer1, 2019-03-09:
Might be out of the scope of this project.  RGB RAM is not interfaced via USB, unlike keyboards, mice, mouse mats, etc.  Instead, RAM module RGB controllers connect via the SMBus interface.  I've got my own project where I'm working to reverse engineer ASUS Aura lighting on motherboards.  The Aura controllers on most motherboards use SMBus as well, and many of the available RGB RAM modules use an Aura-protocol compatible controller.  Corsair RGB RAM does not, but since I already was working on SMBus and RGB RAM I picked up a Corsair Vengeance RGB stick to reverse engineer.  I plan on supporting them in my project and already have the commands figured out for basic color and breathing mode control.

Issue on my project for tracking RGB RAM reverse engineering progress:

https://gitlab.com/CalcProgrammer1/OpenAuraSDK/issues/8#note_148771861

I plan on continuing to pick up other RGB RAM modules as well if I find them cheap on eBay.

tatokis, 2019-03-09:
I have reverse engineered the protocol for the RAM and have also created some software for it. I just haven't published it due to lack of time

esion-net, 2019-03-27:
@tatokis Since I got a new pc with those RAM running on linux - Any plans to publish it anytime soon? :)

bitcharmer, 2023-11-21:
> I have reverse engineered the protocol for the RAM and have also created some software for it. I just haven't published it due to lack of time

It's been 4 years now. Can you please share what you have so someone else can pick this up?

gravityfargo, 2024-05-01:
Closing this in favor of #589 since a specific model was listed.

closed 2024-05-01 by gravityfargo