ckb-next/ckb-next #320: ST100 Support

RobRoyce, 2018-12-27:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

I read in #156 that ST100 support is already enabled, but I'm unable to get it to work on my Mac. Any idea what might be going wrong?

> ckb-next-dev-detect - Wed Dec 26 19:11:04 PST 2018

Kernel: Darwin Roberts-MacBook-Pro-2.local 18.2.0 Darwin Kernel Version 18.2.0: Mon Nov 12 20:24:46 PST 2018; root:xnu-4903.231.4~2/RELEASE_X86_64 x86_64
[... 375 lines cut ...]
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[W] add_hid (usb_mac.c:795): Got unknown SINGLE_EP handle (I: 64, O: 64, F: 0)
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[W] add_hid (usb_mac.c:795): Got unknown SINGLE_EP handle (I: 64, O: 64, F: 0)
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[W] add_hid (usb_mac.c:795): Got unknown SINGLE_EP handle (I: 64, O: 64, F: 0)
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[W] add_hid (usb_mac.c:795): Got unknown SINGLE_EP handle (I: 64, O: 64, F: 0)

GUI log:
Path to settings: "/Users/Rob/Library/Preferences/com.ckb-next.ckb-next.plist"
Looking for old settings in: "/Users/Rob/Library/Preferences/com.ckb.ckb.plist"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/gradient"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/heat"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/invaders"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/pinwheel"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/rain"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/random"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/ripple"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/wave"
objc[410]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fff91d9b210) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x113ad2dc8). One of the two will be used. Which one is undefined.

tatokis, 2018-12-27:
Apparently it hasn't made it to release yet. Here's a fresh build with the latest changes.
http://ckb-next.info/mac/ckb-next_v0.3.2-33-g2fb5ba6_10.11-SDK.dmg

```
3d811e89ddde46b5a9140fb47e617eb5a0268c9b395fc536053701d7366c332b  ckb-next_v0.3.2-33-g2fb5ba6_10.11-SDK.dmg
```

RobRoyce, 2018-12-28:
Awesome, works perfectly now. Thanks!

closed 2018-12-28 by RobRoyce