ckb-next/ckb-next #321: K65 non-RGB is not detected

yqt, 2018-12-27:
Hi, I am switching to macOS recently and trying to get my old K65 non-RGB fully worked on my MBP. 

But cbk-next  shows "No devices connected" with version 0.3.2 installed.(report attached)

I search a lot on Google and find a same issue on https://github.com/mattanger/ckb-next/issues/31. It seems that it is a  issue should not be fixed one year ago. After reading the comments on that issue, I try to apply the patch for it and build a new package. But current code changed a lot and is not suitable for that patch.

[ckb-next-report-1545909754.gz](https://github.com/ckb-next/ckb-next/files/2712361/ckb-next-report-1545909754.gz)

tatokis, 2018-12-27:
Can you please run USB Prober as attached, switch to the `Bus Probe` tab, cilck the `Save Output` button, and attach the generated report to this issue
[USB Prober.zip](https://github.com/ckb-next/ckb-next/files/2713202/USB.Prober.zip)
.

yqt, 2019-01-02:
> Can you please run USB Prober as attached, switch to the `Bus Probe` tab, cilck the `Save Output` button, and attach the generated report to this issue.
> [USB Prober.zip](https://github.com/ckb-next/ckb-next/files/2713202/USB.Prober.zip)

Sorry for the late reply. USB info attached.
[USB Bus Probe.txt](https://github.com/ckb-next/ckb-next/files/2719770/USB.Bus.Probe.txt)

tatokis, 2019-01-05:
Thank you. I will try to get a build for you to test in the upcoming days

tatokis, 2019-01-23:
Apologies. I couldn't figure it out initially, so I put it aside and forgot about it as I got distracted by other issues.

Try this build, see if anything changes and upload a new ckb-next report.

If it doesn't work, the new report should give us more information on how to get it working.
http://ckb-next.info/mac/ckb-next_v0.3.2-g0932a81-k65_1.dmg

```
775849865af0c88ce1f894ee68b9d4f8388f70c651bd88c9823982fa609ab167  ckb-next_v0.3.2-g0932a81-k65_1.dmg
```

Again, apologies for being this late.

yqt, 2019-01-23:
And I got three keyboards listed on setting of keyboard in MacOS with K65 detected by ckb-next.

- Apple internal keyboard/touch pad
- Karabiner VirtualHIDKKeyboard
- Corsair K65 Gaming Keyboard

But only two keyboards listed on that page except `Karabiner VirtualHIDKKeyboard` while K65 is not detected by ckb-next.

tatokis, 2019-01-23:
Also, interesting thing. The log is filled with `[W] hid_kb_translate (keymap.c:471): Got unknown legacy data`

Can you run in a terminal 
```
tar -czf daemonlog.tar.gz /var/log/ckb-next-daemon.log
```
and then upload the resulting daemonlog.tar.gz file?

yqt, 2019-01-24:
Sure. Demon log attached.
[daemonlog.tar.gz](https://github.com/ckb-next/ckb-next/files/2789888/daemonlog.tar.gz)

Thank you.

qdingx, 2020-03-03:
I'm getting the same issue too