ckb-next/ckb-next #322: Dark Core Se support

DarkLogicX, 2018-12-29:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

DarkLogicX, 2018-12-29:
Are there any plans for support for the Corsair Dark Core Se mouse?

I've been using a Scimitar mouse and decided I look the shape of the Dark Core SE better.
After plugging it in (wired, no plans to use wireless) it's able to control the curser, but CKB-next doesn't show a tab for it and it's way to sensitive.

I noticed it's not on the supported list.
https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2715945/ckb-next-dev-detect-report.gz)

ZirconiumX, 2018-12-29:
So, the Dark Core series use a new set of wireless commands, which I tried to implement in the `wireless` branch. But at the moment both main developers have burned out on CKB, so development of major new features like this isn't a very high priority.

The wireless branch is PR #158, and this issue is a duplicate of #101, so I'm closing this.

closed 2018-12-29 by ZirconiumX