ckb-next/ckb-next #324: Mojave don't remember profile settings after restart

hkkho, 2018-12-30:
# Bug report

## General information
  * Mojave don't remember profile settings after restart
### Source
  * pre-built pkg for macOS
## What is the program's current behavior?
  * reload default profile
## What is the expected behavior of the program?
  * Reload previously exported or used profile
## Any other notes & comments?
  * Sometimes it works sometimes it doesn't. Usually don't work on restart.

tatokis, 2018-12-30:
If you click the quit button from the settings tab before rebooting, does it keep the settings?

hkkho, 2018-12-30:
Oh yes it seems if I close it before restarting it remembers the profile. Is there a way to have ckb shutdown itself before macos restart or shutting down?

tatokis, 2018-12-31:
How are you restarting your system?

hkkho, 2018-12-31:
Apple logo -> restart or shutdown

tatokis, 2019-01-07:
This sounds like ckb-next isn't shutting down cleanly and is corrupting the settings. Neither of those should happen, and I'm not sure where to go from here...

hkkho, 2019-01-09:
Yeah just to add some extra information, when I restart and if it bugs out. The profile needs to be imported again, it forgets that it even had profiles to load. I cannot simple switch to the correct profile in the application.

tatokis, 2019-01-31:
Can you try this build? It uses a different way to store the settings, so if this works, then I will switch the whole application to this.

It will not keep any settings from the 0.3.2 build, however it won't delete them either. They store the settings separately, so can always switch back to 0.3.2 if you want your old settings.

http://ckb-next.info/mac/ckb-next_v0.3.2-60-g41a2199_10.11-mac-ini.dmg

```
a71bd394e79b332a5268de617622c36dc860c824e1e904a154df58b6a202f09b  ckb-next_v0.3.2-60-g41a2199_10.11-mac-ini.dmg
```

Please let me know if this fixes the issue.

SSmale, 2019-01-31:
> It will not keep any settings from the 0.3.2 build, however it won't delete them either. They store the settings separately, so can always switch back to 0.3.2 if you want your old settings.

How do I go back?

tatokis, 2019-01-31:
@SSmale You install 0.3.2 https://github.com/ckb-next/ckb-next/releases/download/v0.3.2/ckb-next_v0.3.2.dmg

alexclifford, 2019-02-06:
@tatokis that build fixes the modifier key resetting after sleep issue for me.

shasen1235, 2019-02-18:
Oh boy this finally fixed my issue! I've been using the test build for a week and so far it remembers my profile well. Environment: 10.13.6(17G4015)

closed 2022-03-05 by tatokis