ckb-next/ckb-next #325: Harpoon RGB mouse, can't change color

robo2bobo, 2019-01-01:
# Bug report

## General information

Maybe I did not understand the procedure correctly, but I can't change color on my Harpoon RGB mouse. The same procedure to select a zone and choose a color worked fine for my K55 RGB keyboard.

### Source
Corp johanh/ckb/
ckb-next-0.3.2-2.fc29.x86_64

## What is the program's current behavior?

Here are the steps to reproduce the problem:

1) Select the "HARPOON RGB" tab
2) Click on "New mode..." and create a new mode named "Static"
3) Click on "Change color" and select a dark blue (#00007f)
 ... nothing happens...
5) Click on "Save to hardware"
 ... the Corsair color blinks to blue once and reverts back to default Green.

## What is the expected behavior of the program?

I'd expect the blue color to stay.

## Any other notes & comments?

When I first started ckb-next, I was prompted to upgrade the firmware to the latest version, I accepted to upgrade and the procedure finished without errors.

The device tab says the current firmware is 3.08, with a poll rate of 1ms.

Another interesting thing:
- the "Rainbow" mode makes the LED a constant yellow (no matter what animation I choose).
- the "Breathing" mode makes the LED fade from light blue to dark blue.
- the "Trippy" mode makes the LED a constant purple (no matter what animation I choose).

Thank you.

ZirconiumX, 2019-01-01:
>    4. Click on "Save to hardware"

Can you try unticking "Use DPI Indicator" in the Performance tab?

tatokis, 2019-01-05:
Glad to hear. Happy new year to you too. Closing this in favour of issue #328

closed 2019-01-05 by tatokis