ckb-next/ckb-next #327: Fix default profile bindings

tatokis, 2019-01-02:
The default profiles that come with the gui need to be either modified to map to zones like on the polaris/ST100/K55, or be separated to different ones for each device.

As it is currently, the devices mentioned above just light up white by default, which is not a great first thing to see on an RGB device. Some users get confused and think there's something broken.

Stef500, 2019-01-20:
Hi tatokis, I have the problem. How do I change the color from white to rgb in K55 ? Everything else is working just fine. big kudos on that!

tatokis, 2019-01-20:
In the lighting tab, click on one of the animations on the bottom left.
Then select all 3 zones and click on "set keys" on the bottom left.

Repeat for all animations

Stef500, 2019-01-20:
YOU ROCK! It's awesome. Thx++++++++++++++

jontrodrigues, 2022-09-21:
Hello, I am interested in working on this issue, is it still available?

tatokis, 2022-09-24:
@jontrodrigues Some of the devices (at least the K55) have had their own demo profiles added for some time now, however there are a few more devices that have jarring white LEDs by default.

It is currently a bit of a pain to test, but in order to see how each device shows up, you need to:
1. Quit the GUI
2. Delete `~/.config/ckb-next/` (Note this will delete all your existing profiles, so you might want to back it up the first time.)
3. Open `/dev/input/ckb1/features` as root with a text editor like nano.
4. Change the model to one from this list (the strings enclosed in double quotes). For example if you have an M65 and you want to make it show up as a K70, you'll simply replace `m65` with `k70` and leave everything else as-is. https://github.com/ckb-next/ckb-next/blob/4f274b31656dd5318c13ebdcb727f873b934f05d/src/gui/keymap.cpp#L1367-L1431
5. Restart the GUI.

This will trick it into thinking you have a different device and load the demo profile for it. If you do not delete the config file, then it will not load the demo profiles for that device and keep the previous ones.

The devices/keymaps that have custom profiles can be found here:
https://github.com/ckb-next/ckb-next/blob/2bd1a31552b2bf20a75b941134065bacbe4eccc5/src/gui/kb.cpp#L335-L346