ckb-next/ckb-next #330: Lenovo Legion Y730 - K95 RGB Platinum keyboard not detected

miharkula, 2019-01-04:
[ckb-next-report-1546618035.gz](https://github.com/ckb-next/ckb-next/files/2727615/ckb-next-report-1546618035.gz)
# Bug report

## General information

### Source
build manually using quickinstall script

## Logs & crash reports

Attached report from the app

## What is the program's current behavior?

Lenovo Legion Y730 has a build in keyboard that is being detected as a K95 RGB Platinum in the logs but is not being detected by the application

## What is the expected behavior of the program?

Keyboard is detected

tatokis, 2019-01-04:
@miharkula Can you please post the full output of `lsusb -v` ?
Looks like they aren't using the corsair vendor ID which makes me wonder if this is official or not

tatokis, 2019-01-05:
This definitely looks nothing like a K95 platinum.
Most notably: the product and vendor ids are completely different, with the vendor being ITE and not Corsair.
```
  idVendor           0x048d Integrated Technology Express, Inc.
  idProduct          0xc935 
```
The manufacturer descriptor string is also "Corsai" instead of "Corsair", which is pretty funny.
```
  iManufacturer           1 Corsai
```

The firmware version reported in the usb descriptor makes no sense (13.06).
Do you still have windows/iCUE installed? If so, can you check what firmware version it shows up as, and if it supports firmware updates?

It might be possible to get it to work, as long as they implemented the 0x03 input hid commands, and most of the HID descriptors are the same.

miharkula, 2019-01-05:
Not currently I'll have to reinstall windows temporarily and check.

tatokis, 2019-01-05:
Please get the code from https://github.com/ckb-next/getUsbString, compile it, run it like this, and then post the output.

```
sudo ./getUsbString 0x048d 0xc935 4
```

miharkula, 2019-01-05:
I get an error when I run that. 
`Failed to get string descriptor 4 with -9 (LIBUSB_ERROR_PIPE: Pipe error)`

miharkula, 2019-01-05:
I changed the 4 to a 2 (saw an example from the readme) and got this:
`Read 82 bytes
Raw (hex): 52 03 43 00 6f 00 72 00 73 00 61 00 69 00 72 00 20 00 47 00 61 00 6d 00 69 00 6e 00 67 00 20 00 4b 00 39 00 35 00 20 00 52 00 47 00 42 00 20 00 50 00 4c 00 41 00 54 00 49 00 4e 00 55 00 4d 00 20 00 4b 00 65 00 79 00 62 00 6f 00 61 00 72 00 64 00 
String: Corsair Gaming K95 RGB PLATINUM Keyboard
`

miharkula, 2019-01-05:
I get the error above when I use 4

ZirconiumX, 2019-01-05:
Hmm, perhaps they didn't actually include a protocol string. That would be
funny, if awkward for us.

tatokis, 2019-01-06:
Can you please run `sudo usbhid-dump -m 048d:c935` ?

miharkula, 2019-01-06:
I checked the iCUE software in windows and I do not see where it mentions a firmware version. It shows a plugin version but I did not know if that was the same thing. I will try the above command in a bit after I get Linux back on the laptop.

[image]

miharkula, 2019-01-06:
001:005:001:DESCRIPTOR         1546809417.053502
 06 C2 FF 09 04 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 B1 02 09 04 91 02 C0

001:005:000:DESCRIPTOR         1546809417.054626
 06 89 FF 09 10 A1 01 85 5A 09 01 15 00 26 FF 00
 75 08 95 10 B1 00 C0 06 89 FF 09 07 A1 01 85 07
 09 01 15 00 26 FF 00 75 08 95 BF B1 00 C0 06 89
 FF 09 CC A1 01 85 CC 09 01 15 00 26 FF 00 75 03
 95 20 B1 00 C0 05 01 09 06 A1 01 85 01 75 01 95
 08 05 07 19 E0 29 E7 15 00 25 01 81 02 95 01 75
 08 81 03 95 05 75 01 05 08 19 01 29 05 91 02 95
 01 75 03 91 03 05 07 19 00 2A FF 00 15 00 26 FF
 00 95 0C 75 08 81 00 05 07 19 00 29 DF 15 00 25
 01 95 E0 75 01 81 02 C0 05 0C 09 01 A1 01 85 05
 1A 00 FF 2A 10 FF 16 00 FF 26 10 FF 75 10 95 01
 81 00 C0 05 0C 09 01 A1 01 85 02 19 00 2A 3C 02
 15 00 26 3C 02 75 10 95 01 81 00 C0 05 01 09 0C
 A1 01 85 03 15 00 25 01 09 C6 95 01 75 01 81 06
 75 07 81 03 C0 05 01 09 80 A1 01 85 04 19 81 29
 83 15 00 25 01 95 08 75 01 81 02 C0

lrojas94, 2019-01-10:
Got the same Laptop so I'm also interested to get it to work. Note that when configuring the keyboars through windows iCUE, it is bundled with more than just the keyboard. There's a 2 RGB lights under the chasis that can be customized there, as well as one in the back of the laptop that is also toggled by that. The keyboard might be the same, but there's more lighting to control, which might be the reason for the discrepancies in the product/model Ids

lrojas94, 2019-01-13:
@tatokis any news on this? I'll help in whatever I can

tatokis, 2019-01-13:
@lrojas94 Apologies for missing you on IRC, I was asleep.
That said, I had a look and the protocol for this is vastly different to the rest of the corsair devices, so that is going to be a bit of a pain.

CUE also seems to spam RGB packets to it even when there is no activity (the lights are off), so that's not nice. If you can drop back in on IRC, I can ask you to try a few things.

tatokis, 2019-01-13:
Can you please run `cat /sys/devices/virtual/dmi/id/product_name` ?

lrojas94, 2019-01-13:
Response I got from that is:
```
81HD
```

joro1881, 2019-01-21:
Greetings all. Another Y730 customer using Ubuntu, interested in the issue resolving. I am almost online 24/7 So poke me here if I can assist or provide information. 

I also got '81HD' as output from the cat... product_name comand.

tatokis, 2019-01-21:
@joro1881 is it the one with the numpad?

joro1881, 2019-01-21:
@tatokis Nope the 15''

miharkula, 2019-01-22:
When I use Fn + space 4 times it just turn back off again.

miharkula, 2019-01-22:
Checked again when I got home and noticed something I hadn't before at work..probably due to the ceiling lights making it hard to notice.

'Fn' + space rotations:
1 - Turns keyboard, exhaust ports and logo to blue
2 - Rainbow from key effect, exhaust ports and logo rotate through colors
3 - Keyboard off, exhaust ports and logo blue
4 - All lights off

@joro1881 What did you mean by increase / decrease buttons? Are you referring to the two keys under M1 and M2 that change the brightness of the leds?

tatokis, 2019-01-23:
> Nope, the 15'' You think it might has different id of the keyboard ? The correct Corsair one ?

They both have the same ID and the same new protocol.

> the ckb-next functionality is working :D

No, it's not. It's whatever is built in to the firmware of your device.

cattalinss, 2019-01-24:
Hi there! I am also the (happy so far) owner of this laptop also (15 inch version) and I would be very happy to be able to control the rgb keyboard in Linux. Please let me know if I can help somehow. I use Ubuntu 18.04 and for now the ckb interface is telling me "no devices connected".
Thank you

lrojas94, 2019-02-03:
Hey @tatokis, sorry I went MIA suddenly... Work got the best out of me on the last 2 weeks. Can you remind me the filter for Wireshark to get the correct packages? Working on the mapping, but having a hard time due to them missing.

lrojas94, 2019-02-11:
@tatokis any updates worth mentioning? curious as if you even saw the packages 😄

tatokis, 2019-02-11:
@lrojas94 sorry, I got busy with other things. I'll download it now. 

That said, there is a slight issue I've been thinking about.
CUE doesn't handle key mapping on windows. 

It talks to the keyboard using URB Controls, which do not require exclusive access to the device. If we were to do the same, we'd only be able to send rgb data, and like CUE, not mess with any bindings. 

However, then I'm not sure how things like interactive animations would work, since we won't be able to read from the keyboard. 

If we were to try to get exclusive access, we'd need to implement the hid protocol specific to this device, as ckb is not a full blown hid driver, and that might be a challenge as it is probably wildly different to what corsair uses. 

And all of that implies that the hid packets come from the same device, and it's not just an rgb controller with a separate keyboard, which is what I'm afraid is probably the case.

lrojas94, 2019-02-11:
If it is possible to at least control the key colors - even though animations are out of the picture - I think it wouldn't be too bad of a tradeoff (I mean, is either that or no support right?). Not sure if worth your time tho. 

As I explained before, the work around I'm using is leaving Windows in Hibernate (this for some reason will leave whatever config. you've left onto the keyboard ON) and going into Ubuntu afterwards. Animations are not kept at all, but hey, can't win them all 😄

joro1881, 2019-02-17:
"[tatokis](/tatokis) referenced this issue [ an hour ago](#ref-issue-411169014)" 

So you won't have the time to work on solution with this :( or there will be one in future updates, or it's totally problem with Lenovo scrapping the keyboard?

tatokis, 2019-02-17:
It will happen eventually. Working on this is currently blocked until I get the m95 branch merged, as that contains code to send URB Controls in a centralised manner.

@lrojas94 I meant interactive, like typing and having a ripple show up around the key you pressed.
Assuming you have another keyboard plugged in, can you try running
```
sudo usbhid-dump -d 048d:c935 -e stream
```
pressing the keys <kbd>a</kbd> <kbd>b</kbd> and <kbd>c</kbd> on the **laptop** keyboard and then pasting the output?

miharkula, 2019-02-17:
@tatokis I tried that and this was what I get (command typed with external keyboard and abc on the laptop):
```
jason@legion:~$ sudo usbhid-dump -d 048d:c935 -e stream
[sudo] password for jason: 
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:007:001:Interrupt transfer failed
abc
```

lrojas94, 2019-02-18:
```
sudo usbhid-dump -d 048d:c935 -e stream 
[sudo] password for luis: 
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:005:001:Interrupt transfer failed
abc
001:005:000:Interrupt transfer timed out
No more interfaces to dump

```

Basically the same thing really.

tatokis, 2019-02-18:
What about `sudo usbhid-dump -d 048d:c936 -e stream` ?

lrojas94, 2019-02-18:
```
sudo usbhid-dump -d 048d:c936 -e stream
[sudo] password for luis: 
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:004:000:STREAM             1550497686.287482
 01 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:004:000:STREAM             1550497686.363453
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:004:000:STREAM             1550497686.367441
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:004:000:STREAM             1550497688.195485
 01 00 00 05 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:004:000:STREAM             1550497688.243499
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:004:000:STREAM             1550497688.247461
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:004:000:STREAM             1550497689.167518
 01 00 00 06 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:004:000:STREAM             1550497689.223408
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

001:004:000:STREAM             1550497689.227466
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00
```

This is the output while using `a,b,c` as proposed. Make sure to use another keyboard to avoid any noise.

lrojas94, 2019-03-02:
@tatokis any updates around this?

tatokis, 2019-03-06:
I've been preparing for the next release which will be coming out in the next few days. Only M95 support (which has the URB control functions needed for the Y730) is left, which I am going to be done polishing up really soon.

After the next release I will start working on this.

lrojas94, 2019-03-07:
:+1: Please let me know if I can help you in any way!

Blendman974, 2019-03-10:
Interested in buying the legion Y740 wich should have the same keyboard. Any chances to fully support all of the keyboard features ? (Animations, macros and ambiant lights)

lrojas94, 2019-03-22:
@tatokis Any updates? Saw the new update dropping a few days ago, and since release notes included M95 Legacy support, wanted to see if there was an update or I should just let it go.

tatokis, 2019-04-03:
@lrojas94 I did start working on it. However it requires a few changes, that while easily done on linux, are a pain to do on mac.

Currently the daemon assumes each device uses the 0x1b1c corsair vendor id, which is not the case here. For the moment I might not fix mac support, so that we can at least get it working on linux.

That said, this is what I have so far. https://github.com/ckb-next/ckb-next/commit/977a8d83d1b4a2194578068af11c40453c4882c6
Need to start implementing the vtable commands, which should be relatively easy.

tatokis, 2019-04-06:
Can someone try the code from the [legion](https://github.com/ckb-next/ckb-next/tree/legion) branch?
Please run the daemon from the terminal, and paste its log here.

If it doesn't crash, you can try the following commands:

`echo active > /dev/input/ckb1/cmd` (assuming ckb1 is the keyboard) which should switch it to software mode.
`echo rgb ff00ff > /dev/input/ckb1/cmd` should set the colour for most keys.
`echo idle > /dev/input/ckb1/cmd` should revert the keyboard back to hardware mode.

Additionally, assuming this all works, can someone test if the brightness button on the keyboard itself works while in software mode?

lrojas94, 2019-04-06:
Not home atm, but will try this out and let you know. Also, my hardware knowledge is pretty limited, but whatever you're willing to guide me through I'll do :) thank you!

T3CHKOMMIE, 2019-04-06:
I'll give it a shot this afternoon as well

-jules

T3CHKOMMIE, 2019-04-06:
I pulled the legion branch and did another quick install, not sure how to fireoff the daemon. pointers on how to kick the darmon off from terminal. I am pretty ok with CLI, but not sure where the daemon is or its parameters.

ZirconiumX, 2019-04-06:
Just run `sudo ckb-next-daemon`.

T3CHKOMMIE, 2019-04-06:
im just geting permissions denied even with sudo

jules@LinuxLegion:~/ckb-next-legion/ckb-next$ sudo ckb-next-daemon
ckb-next: Corsair RGB driver 0.4.0-5-g11af88d
[I] Root controller ready at /dev/input/ckb0

jules@LinuxLegion:~$ echo active > /dev/input/ckb0/cmd
bash: /dev/input/ckb0/cmd: Permission denied
jules@LinuxLegion:~$ sudo !!
sudo echo active > /dev/input/ckb0/cmd
bash: /dev/input/ckb0/cmd: Permission denied

tatokis, 2019-04-06:
A. you're not running the code from the legion branch
B. it's ckb1, not ckb0. The reason you don't have a ckb1 is because you're not running the code from the branch.

T3CHKOMMIE, 2019-04-06:
ah, damnit. looks like my git clone pulled the master for some reason.
 lemme try again.

tatokis, 2019-04-06:
You need to `git checkout legion` after you clone.

T3CHKOMMIE, 2019-04-06:
alright, 

jules@LinuxLegion:~/ckb-next-legion/ckb-next$ git branch
* legion
  master
jules@LinuxLegion:~/ckb-next-legion/ckb-next$ sudo ckb-next-daemon
ckb-next: Corsair RGB driver 0.4.0-5-g11af88d
[I] Root controller ready at /dev/input/ckb0

jules@LinuxLegion:~$ sudo echo active > /dev/input/ckb0/cmd
bash: /dev/input/ckb0/cmd: Permission denied

let me know if im doing something wrong still. when the daemon is off i have no ckb input devices. not sure how i would have a ckb1.

tatokis, 2019-04-06:
>0.4.0-5-g11af88d

That's still master.

Try `mkdir build && cd build && cmake .. & make -j4 daemon`
and then `sudo ./bin/ckb-next-daemon`

T3CHKOMMIE, 2019-04-06:
ah looks like i pulled the branch but didnt install it. quick install has it working with ckb1

standby

T3CHKOMMIE, 2019-04-06:
jules@LinuxLegion:~/ckb-next-legion/ckb-next$ sudo ckb-next-daemon
ckb-next: Corsair RGB driver 0.4.0-9-g28505ae
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming K95 RGB PLATINUM Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Setup finished for /dev/input/ckb1

This look about right?

T3CHKOMMIE, 2019-04-06:
setup finished on ckb1...

jules@LinuxLegion:~$ sudo echo active > /dev/input/ckb1/cmd

[E] os_usbsend_control (via device_laptop.c:17):  Broken pipe, res = 0xffffffff

jules@LinuxLegion:~$ echo rgb ff00ff > /dev/input/ckb1/cmd

[E] os_usbsend_control (via device_laptop.c:110):  Broken pipe, res = 0xffffffff
[E] os_usbsend_control (via device_laptop.c:110):  Protocol error, res = 0xffffffff
[E] os_usbsend_control (via device_laptop.c:110):  Broken pipe, res = 0xffffffff

tatokis, 2019-04-06:
Can you check the last few lines in `dmesg`? There are probably some error messages that might give more information as to why this fails

T3CHKOMMIE, 2019-04-06:
let me know if this isn't what you are asking for. Im not fluent in dmesg but i dont see any errors.

[ 5651.597622] i2c_hid i2c-ELAN0504:01: i2c_hid_get_input: incomplete report (14/65535)
[ 5651.604571] i2c_hid i2c-ELAN0504:01: i2c_hid_get_input: incomplete report (14/65535)
[ 5652.650888] input: Corsai Corsair Gaming K95 RGB PLATINUM Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:048D:C935.0013/input/input73
[ 5652.708216] input: Corsai Corsair Gaming K95 RGB PLATINUM Keyboard Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:048D:C935.0013/input/input74
[ 5652.708579] input: Corsai Corsair Gaming K95 RGB PLATINUM Keyboard Wireless Radio Control as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:048D:C935.0013/input/input75
[ 5652.708907] input: Corsai Corsair Gaming K95 RGB PLATINUM Keyboard System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:048D:C935.0013/input/input76
[ 5652.709621] hid-generic 0003:048D:C935.0013: input,hiddev1,hidraw2: USB HID v1.10 Keyboard [Corsai Corsair Gaming K95 RGB PLATINUM Keyboard] on usb-0000:00:14.0-6/input0
[ 5652.711451] hid-generic 0003:048D:C935.0014: hiddev2,hidraw3: USB HID v1.10 Device [Corsai Corsair Gaming K95 RGB PLATINUM Keyboard] on usb-0000:00:14.0-6/input1

T3CHKOMMIE, 2019-04-06:
this looks more helpful:

[ 5865.877078] i2c_hid i2c-ELAN0504:01: i2c_hid_get_input: incomplete report (14/65535)
[ 5865.884028] i2c_hid i2c-ELAN0504:01: i2c_hid_get_input: incomplete report (14/65535)
[ 5894.889864] usb 1-6: usbfs: USBDEVFS_CONTROL failed cmd ckb1 usb rqt 64 rq 9 len 192 ret -71
[ 5910.485394] usb 1-6: usbfs: USBDEVFS_CONTROL failed cmd ckb1 usb rqt 64 rq 9 len 192 ret -71
[ 5910.486033] usb 1-6: usbfs: USBDEVFS_CONTROL failed cmd ckb1 usb rqt 64 rq 9 len 192 ret -71

tatokis, 2019-04-06:
Run `git pull` in the repository and try again with the latest code

T3CHKOMMIE, 2019-04-06:
Awesome! I can now change the color of the backlight! backlight brighness button works in software mode. 

The only thing that seems to not be working is the BL behind the numpad (I have the 17 in legion), right ctl and del buttons are not backlit. Im guess that's just a few tweaks from working?

what else would you like me to test?

lrojas94, 2019-04-06:
Here on the 15 in. Tested with the latest branch, I can change the colors using the code above. Tried a few (white, purple, etc). The Side-Macro keys don't turn-on (brightness is one of these), but they are functional - allowing to properly control the brightness

lrojas94, 2019-04-06:
Note that - aside from the macro - The program accurately controls *every* single led on the 15 in, including the Power Button led, as well as the back-logo led & side ones.

[46 more replies]