ckb-next/ckb-next #338: K70 Rapidfire issues with breathing mode.

Raymondms99, 2019-01-06:
Hi, I've been using CKB-Next for about 6 months now, no change to the setup. I was playing around with a different mode because I've always had it on Rainbow. When I clicked over to the breathing mode the keyboard flashed then CKB told me no devices connected. I also have a m65 RGB plugged in. I've reinstalled, keyboard is on 1 on the bios switch. Any ideas on how I might be able to reset the program manually?
[ckb-next-report-1546765525.gz](https://github.com/ckb-next/ckb-next/files/2730303/ckb-next-report-1546765525.gz)

tatokis, 2019-01-08:
I'm not sure I have seen this before. Have you tried rebooting/unplugging and replugging the keyboard?

gravityfargo, 2024-05-01:
This issue has been marked as stale due to age and/or lack of activity. 

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-01 by gravityfargo