ckb-next/ckb-next #339: Middle Mouse click not working

Hellbringher, 2019-01-07:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?
middle mouse click for scimitar pro doesn't work...

## What is the expected behavior of the program?
is possible to fix it using a macro until next version release? i'm using 0.3.2

## Any other notes & comments?

---

# Something else

Hellbringher, 2019-01-07:
I'm with problems with middle mouse click for Scimitar Pro... when click, nothing happens! Is it possible to assign a macro to make click work until a future version that fix it?

tatokis, 2019-01-07:
Middle click is supposed to be working. Make sure your middle button is assigned to `Standard button: Middle` in the binding tab.
[image]

Can you also please click the "Generate Report" button in the Settings tab, and then attach the generated report here?

Hellbringher, 2019-01-07:
Config was like your screenshot... attached the Report generated.

---
Paulo Campoy
+55 12 98154-5666
501st Commanding Officer (CO) - Brazil: Brazilian Division Garrison
visit my 501st profiles at:
Stormtrooper  -  http://goo.gl/khdYpc
Biker Scout  -  http://goo.gl/biQXEP

tatokis, 2019-01-07:
I don't see anything attached. I think you need to use the github website to attach a file.

Hellbringher, 2019-01-07:
I found this post too...

https://github.com/4ian/GDevelop/pull/841

---
Paulo Campoy
+55 12 98154-5666
501st Commanding Officer (CO) - Brazil: Brazilian Division Garrison
visit my 501st profiles at:
Stormtrooper  -  http://goo.gl/khdYpc
Biker Scout  -  http://goo.gl/biQXEP

Hellbringher, 2019-01-07:
using the githup page now to attach
[ckb-next-report-1546830744.gz](https://github.com/ckb-next/ckb-next/files/2731293/ckb-next-report-1546830744.gz)

Hellbringher, 2019-01-07:
Where I use middle click? when using google chrome or firefox to open the link on a new tab...

tatokis, 2019-01-18:
Run `sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist`, unplug and then replug your mouse.
Does the middle click now work?

If it still doesn't work, then it's a hardware problem and we can do nothing about it.
You can then run `sudo launchctl load /Library/LaunchDaemons/org.ckb-next.daemon.plist` to restart the ckb-next driver.

Hellbringher, 2019-01-18:
Hi Tasos,

Doesn't work... but I did a test! I got one sit that capture mouse click
and when I click on wheel nothing happens. None event raised.
As you told or last firmware disabled the wheel click or mine mouse was
damaged.

Thanks for your time and attention,
---
Paulo Campoy
+55 12 98154-5666
501st Commanding Officer (CO) - Brazil: Brazilian Division Garrison
visit my 501st profiles at:
Stormtrooper  -  http://goo.gl/khdYpc
Biker Scout  -  http://goo.gl/biQXEP

noshky, 2020-07-13:
Same trouble here. Middle mouse button not giving any event to the system.
As [suggested here](https://help.ubuntu.com/community/MouseCustomizations):
```
sudo apt-get install xautomation
# and
xev
```
doesn't show anything when clicking the middle button.

closed 2022-10-03 by tatokis