ckb-next/ckb-next #342: K70 LUX non-RGB not being able to update firmware

jencinarh, 2019-01-07:
# Bug report

## General information
I've bought the K70 lux non-rgb version. I've plugged in ubuntu 16.04 and it does not allow me to update its firmware.

When I open ckb-next GUI it detects the keyboard for a second and then disconnects/no longer detects the keyboard for another second and then loops over and over again.

I've been reading the logs and it's throwing some errors :  
- Jan 07 13:02:36.191459 RBPMI-131 ckb-next-daemon[822]: [E] os_setupusb (usb_linux.c:597): Unable to read endpoint count from udev, assuming 0
- Jan 07 13:07:53.712734 RBPMI-131 ckb-next-daemon[4288]: [E] loadrgb_kb (led_keyboard.c:314): Bad input header

### Source
  * built manually using quickinstall script

## Logs & crash reports
Attached `ckb-next-dev-detect-report.gz` 

## What is the program's current behavior?
The current behavior is that the software its not able to update the firmware of the keyboard.

## What is the expected behavior of the program?
The software should recognize the keyboard and allow the firmware update

## Any other notes & comments?
I'm experiencing issues when trying to use the keyboard in a non-bios mode. 
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2733026/ckb-next-dev-detect-report.gz)

tatokis, 2019-01-07:
Are any lights on it flashing all the time?

jencinarh, 2019-01-07:
Not when it's not in bios mode

tatokis, 2019-01-07:
Okay, it appears you are running a really really old version of the firmware, which doesn't play nice with linux. Please join us on our IRC room and we can assist you. http://webchat.freenode.net/?channels=%23ckb-next&uio=d4

closed 2019-01-07 by jencinarh

Odzinic, 2020-06-10:
I am experiencing this same issue. Was a solution found for this?

jencinarh, 2020-06-11:
It seemed at the beginning it was fixed by updating the bios. However,
shortly after it started failing again.
Thus, I returned the keyboard. I recommend you to return it asap and get
another K70 (or other keyboard).