ckb-next/ckb-next #345: Ironclaw RGB

ZirconiumX, 2019-01-08:
Corsair have announced [the Ironclaw RGB](https://m.hexus.net/tech/reviews/peripherals/125795-corsair-ironclaw-rgb/).

Looks like we'll need to set USES_FILE_HWSAVE, and let the fancy three hardware animation profiles go to waste.

davidhiendl, 2019-03-15:
Is there any support for the Ironclaw RGB yet? (Not talking about LEDs, mostly interested in DPI/Buttons?)
At the moment it seems like it is not detected.

ZirconiumX, 2019-03-15:
No, because we need a packet trace of the Ironclaw under CUE to work out what it's doing. You're welcome to be the person who gives us that.

davidhiendl, 2019-03-15:
Sure, happy to help. No clue how to do that under windows tho, would need some basic instructions.

ZirconiumX, 2019-03-15:
If you join us in IRC by pressing the blue Freenode button in the readme,
we can guide you and work much faster.

ZirconiumX, 2019-05-16:
@davidhiendl Could you help us finish support for the Ironclaw?

davidhiendl, 2019-05-16:
I completely forgot about this, gonna take a look at it this weekend and fix the code. I could use some help with the overlay graphics (I'm not an artist).

ZirconiumX, 2019-05-16:
You mean the art [I just added](https://github.com/ckb-next/ckb-next/blob/ironclaw/src/gui/resources/overlay_ironclaw.png)?

davidhiendl, 2019-05-16:
Yes, I see you already merged my commit - should I still make the changes and fix the button?

ZirconiumX, 2019-05-16:
Sure, I'll merge them in again.

joeknock90, 2019-05-30:
I just got one of these. Let me know if there's a way I can help get this integrated.

tatokis, 2019-05-30:
Get the code from the `ironclaw` branch https://github.com/ckb-next/ckb-next/archive/ironclaw.zip , run quickinstall, start the GUI, press the button that switches profiles on your mouse, and then generate and attach a report.

joeknock90, 2019-05-30:
Alright I've got the ironclaw branch installed, but I don't know what you mean by button that switches profiles on the mouse. The mouse does not show up in the gui still. Keyboard still showing up fine.

joeknock90, 2019-05-30:
I am an idiot and I think i realize what you mean now. Here's a report.

tatokis, 2019-05-30:
You have the wireless version of the mouse, which uses an entirely different and unsupported protocol. There probably won't be support for it for some time.

The wired one (which this issue is about) uses the old protocol that most other corsair peripherals use, which is easy for us to add support for. Unfortunately there isn't much you can do, other than figuring it out and implementing it yourself.

joeknock90, 2019-05-30:
Gotcha. Unfortunate. But I understand. I'll probably end up returning it in that case.

andrewgrasman, 2019-06-01:
@tatokis I have the wired version of the Ironclaw I hit the two buttons that change the sensitivity next to middle click, hope this report helps(I ran the quickinstall script)
[ckb-next-report-1559349926.gz](https://github.com/ckb-next/ckb-next/files/3243229/ckb-next-report-1559349926.gz)

tatokis, 2019-06-02:
@andrewgrasman Thank you, that does in fact help. Can you tell me a bit more about the function of those buttons? For example, do both change the DPI up and down only? Or does one switch between internal profiles.

I know nothing about this mouse, and based on the code @davidhiendl wrote, they don't seem to just be "DPI up" and "DPI down".

andrewgrasman, 2019-06-03:
@tatokis No, it seems like there are three profiles with the top center button(just below the scroll wheel) and the lower center button(just below the previous one) cycles through three dpi settings.

tatokis, 2019-06-19:
@andrewgrasman Making sure ckb-next-daemon isn't running, can you run `sudo usbhid-dump -d 1b1c:1b5d` and paste the output?

davidhiendl, 2019-06-20:
@tatokis This mouse has no DPI up/down, it has a DPI cycle and Profile Cycle button on the top of the mouse. I couldn't figure out how to make profile cycle work, only DPI cycle.

davidhiendl, 2019-06-20:
@tatokis Here is the usbhid-dump:
```
002:004:001:DESCRIPTOR         1561029920.572595
 06 C2 FF 09 04 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 B1 02 09 02 81 02 09 04 91 02 C0

002:004:000:DESCRIPTOR         1561029920.573094
 05 01 09 02 A1 01 85 01 09 01 A1 00 05 09 19 01
 29 20 15 00 25 01 95 20 75 01 81 02 05 01 09 30
 09 31 16 01 80 26 FF 7F 75 10 95 02 81 06 09 38
 15 81 25 7F 75 08 95 01 81 06 C0 C0 05 0C 09 01
 A1 01 85 02 05 0C 19 00 2A 3C 02 15 00 26 FF 0F
 75 10 95 02 81 00 C0 06 C1 FF 09 02 A1 01 85 03
 15 00 26 FF 00 95 3F 75 08 09 02 81 02 C0 06 C2
 FF 09 03 A1 01 85 0E 15 00 26 FF 00 95 3F 75 08
 09 03 81 02 C0 05 01 09 06 A1 01 85 05 05 07 19
 E0 29 E7 15 00 25 01 75 01 95 08 81 02 19 00 29
 97 15 00 25 01 75 01 95 98 81 02 C0
```

tatokis, 2019-06-20:
@davidhiendl this is a bit confusing. Can you run `ckb-next-daemon` in a terminal, make sure the GUI is running and has detected the mouse, and then press the button that doesn't work.

Do you see anything in the daemon log? (Unknown data messages)

If not, open another terminal, `cat /dev/input/ckb1/notify0` and press the button again. It should show you what the daemon thinks the key is.

davidhiendl, 2019-06-20:
@tatokis I just ran the daemon with sudo, started the GUI and clicked the button and got no additional log entries. Also running `cat /dev/input/ckb1/notify0` with and without root yielded no results.

tatokis, 2019-06-20:
Try notify1 instead maybe

davidhiendl, 2019-06-20:
That worked, but it shows only input for all buttons except the profile button. Profile button still does nothing.

tatokis, 2019-06-20:
Can you compile the daemon with `-DDEBUG_USB_INPUT=1` and then run it locally? (Without installing it system wide because it'll flood your logs.)
Something like `mkdir build && cd build && cmake .. -DDEBUG_USB_INPUT=1 && make daemon && sudo ./bin/ckb-next-daemon`

Once you do that and you have the GUI running, try pressing the button and record what the packet that gets sent is.

expired00, 2019-07-16:
I am having troubles getting the two buttons below the mouse wheel to work consistently when setting them to change dpi up/down. I'm happy to provide logs, or any assistance I can - just let me know what I need to do.

nmenon, 2019-09-17:
@tatokis is there a todo list for us to fixup to get this branch merged to master? How can we help?

tatokis, 2019-09-17:
The scancode for the profile button needs to be added to the daemon so that the button works.

nmenon, 2019-09-17:
@tatokis I merged master to branch ironclaw and followed your instructions in https://github.com/ckb-next/ckb-next/issues/345#issuecomment-504079293
Following https://www.corsair.com/corsairmedia/sys_master/productcontent/IRONCLAW_RGB_QSG.pdf (button D):
(NOTE: I have two corsair products on my desktop)
$ sudo ./bin/ckb-next-daemon
ckb-next: Corsair RGB driver 0.4.1-16-gdb4e4434244e
[I] Root controller ready at /dev/input/ckb0
[I] Connecting CORSAIR IRONCLAW RGB Gaming Mouse at /dev/input/ckb2
[I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb2
[.... SNIP...]
[I] Setup finished for /dev/input/ckb1
[I] ckb2 Input Recv: 01 00 00 00 00 00 00 01 00 00 
[I] ckb2 Input Recv: 03 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb2 Input Recv: 02 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 02 from endpoint 81
[I] ckb2 Input Recv: 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81
[I] ckb2 Input Recv: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 
[I] ckb2 Input Recv: 01 00 00 00 00 01 00 00 00 00 
[I] ckb2 Input Recv: 03 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb2 Input Recv: 02 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 02 from endpoint 81
[I] ckb2 Input Recv: 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81
[I] ckb2 Input Recv: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input Recv: 01 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 

[I] ckb1 Input Recv: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb2 Input Recv: 03 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb2 Input Recv: 02 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 02 from endpoint 81
[I] ckb2 Input Recv: 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81
[I] ckb2 Input Recv: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input Recv: 01 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input Recv: 01 01 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
^C
[I] Caught signal 2
[I] ckb1 Input Recv: 01 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c[I] ckb1 Input Recv: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] Disconnecting /dev/input/ckb1
[I] Stopping input thread for /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Disconnecting /dev/input/ckb2
[I] ckb2 Input Recv: 02 00 00 00 00 
[I] Removed device path /dev/input/ckb2
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 02 from endpoint 81
[I] Closing root controller
[I] Removed device path /dev/input/ckb0

nmenon, 2019-09-17:
More controlled test: button references based off https://www.corsair.com/corsairmedia/sys_master/productcontent/IRONCLAW_RGB_QSG.pdf

Button F press:
[I] ckb2 Input Recv: 05 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81
Button F release:
[I] ckb2 Input Recv: 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81

Button G press:
[I] ckb2 Input Recv: 05 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81
Button G release:
[I] ckb2 Input Recv: 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81

Button A press:
[I] ckb2 Input Recv: 01 01 00 00 00 00 00 00 00 00 
Button A release:
[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 

Button B press:
[I] ckb2 Input Recv: 01 02 00 00 00 00 00 00 00 00 
Button B release:
[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 

scroll C up:
[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 01 

scroll C down:
[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 ff 

scroll C press:
[I] ckb2 Input Recv: 01 04 00 00 00 00 00 00 00 00 

scroll C release:
[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 

Button D press:
[I] ckb2 Input Recv: 03 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

Button D release:
[I] ckb2 Input Recv: 02 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 02 from endpoint 81
[I] ckb2 Input Recv: 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81
[I] ckb2 Input Recv: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 

Button E press:
[I] ckb2 Input Recv: 03 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

Button E release:
[I] ckb2 Input Recv: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

tatokis, 2019-09-17:
I am going to guess the F and G buttons don't work either...

nmenon, 2019-09-17:
@tatokis - correct, with the branch merged to master.

nmenon, 2019-09-17:
[image] <- is that how it is supposed to look?

nmenon, 2019-09-17:
@tatokis not exactly what is going on.. I re-built everything (git clean -fdx and mkdir build && cd build && cmake .. -DDEBUG_USB_INPUT=1 && make daemon && sudo ./bin/ckb-next-daemon) .. and this time I moved the mouse off the mouse pad to get good data.

Corrections below: (F and G)

> More controlled test: button references based off https://www.corsair.com/corsairmedia/sys_master/productcontent/IRONCLAW_RGB_QSG.pdf
>
> Button F press:
> [I] ckb2 Input Recv: 05 00 00 00 00 00 01 00 00 [...]

[I] ckb2 Input Recv: 01 10 00 00 00 00 00 00 00 00 

> Button F release:
> [I] ckb2 Input Recv: 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> [E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81
> 

[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 

> Button G press:
> [I] ckb2 Input Recv: 05 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> [E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81

[I] ckb2 Input Recv: 01 08 00 00 00 00 00 00 00 00 

> Button G release:
> [I] ckb2 Input Recv: 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> [E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 05 from endpoint 81
> 

[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 

> Button A press:
> [I] ckb2 Input Recv: 01 01 00 00 00 00 00 00 00 00

[I] ckb2 Input Recv: 01 01 00 00 00 00 00 00 00 00 

> Button A release:
> [I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00

[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 

> 
> Button B press:
> [I] ckb2 Input Recv: 01 02 00 00 00 00 00 00 00 00

[I] ckb2 Input Recv: 01 02 00 00 00 00 00 00 00 00 

> Button B release:
> [I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00

[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 

> 
> scroll C up:
> [I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 01
> 

[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 01 

> scroll C down:
> [I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 ff

[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 ff 

> 
> scroll C press:
> [I] ckb2 Input Recv: 01 04 00 00 00 00 00 00 00 00
[I] ckb2 Input Recv: 01 04 00 00 00 00 00 00 00 00 

> 
> scroll C release:
> [I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00

[I] ckb2 Input Recv: 01 00 00 00 00 00 00 00 00 00 

> 
> Button D press:
> [I] ckb2 Input Recv: 03 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 

[I] ckb2 Input Recv: 03 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

> Button D release:
> [I] ckb2 Input Recv: 02 00 00 00 00
> [E] process_input_urb (keymap.c:355): Unknown mouse data received in input thread 02 from endpoint 81
> [I] ckb2 Input Recv: 05 00 00 00 00 00 00 [...]

[I] ckb2 Input Recv: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

> 
> Button E press:
> [I] ckb2 Input Recv: 03 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 

[I] ckb2 Input Recv: 03 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

> Button E release:
> [I] ckb2 Input Recv: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

[I] ckb2 Input Recv: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

tatokis, 2019-09-17:
The last capture was most likely done without the GUI running, which is why you are seeing these differences.

nmenon, 2019-09-17:
> The last capture was most likely done without the GUI running, which is why you are seeing these differences.
Yes, indeed, no GUI.

tatokis, 2019-09-17:
I have rebased the ironclaw branch and commited a possible fix. Please test it https://github.com/ckb-next/ckb-next/commit/0b141c0cff7c5346dfd285ba4802f1b5bd4e8e39

nmenon, 2019-09-17:
@tatokis thanks. seems to work for me:

$ cat /dev/input/ckb2/model 
CORSAIR IRONCLAW RGB Gaming Mouse
$ cat /dev/input/ckb2/
cmd        dpi        features   fwversion  layout     model      notify0    notify1    pollrate   productid  serial     
$ cat /dev/input/ckb2/fwversion 
0324
$ cat /dev/input/ckb2/notify1 

F:

key +mouse5
key -mouse5

G:

key +mouse4
ey -mouse4

A:

key +mouse1
key -mouse1

B:

key +mouse2
ey -mouse2

Wheel:
key +wheelup
key -wheelup
n
key -wheeldn
se3

key +mouse3
cat /dev/input/ckb1/key -mouse3
key +mouse3
cat /dev/input/ckb1/3

D:

key +profswitch

E:
ey +dpiup
key -dpiup
ey +dpiup
ey -dpiup
key +dpiup
key -dpiup
+dpiup
key -dpiup
key +dpiup
ey -dpiup
key +dpiup
key -dpiup
^C
$

nmenon, 2019-09-17:
* tested with GUI, i had to set the DPI binding to (cycle up)
only two improvements we could have:
a) gui indicator for wheel and logo -> not sure if we can do better representation
b) save to hardware

tatokis, 2019-09-17:
Is there a wheel light?
From what I understand, there's a logo light and the 3 mode/dpi ones on the left side.

nmenon, 2019-09-17:
@tatokis  yes. wheel light, logo light and the dpi light all work fine..
[image]
[image]
my only crib is the icons dont match up with the mouse.. little non-intutive, however, "mouse-over" hint does explain what it is.
[image]

tatokis, 2019-09-17:
I am trying to fix the GUI, that's why I am asking.
The corsair website only mentions 1 RGB zone
https://www.corsair.com/eu/en/Categories/Products/Gaming-Mice/FPS-Fast-Action-Mice/IRONCLAW-RGB-FPS-MOBA-Gaming-Mouse/p/CH-9307011-EU#tab-tech-specs

nmenon, 2019-09-17:
> I am trying to fix the GUI, that's why I am asking.
> The corsair website only mentions 1 RGB zone
> https://www.corsair.com/eu/en/Categories/Products/Gaming-Mice/FPS-Fast-Action-Mice/IRONCLAW-RGB-FPS-MOBA-Gaming-Mouse/p/CH-9307011-EU#tab-tech-specs

Definitely two (wheel and logo). at least on https://www.amazon.com/dp/B07KKZB9LM/

nmenon, 2019-09-17:
@tatokis since I cant upload a video, gif instead..
[image]

tatokis, 2019-09-17:
I have pushed some changes. The issues should be fixed now.
Can you create a new profile and test it there? Also please try rebinding the forward and backward buttons and make sure they are not swapped.

nmenon, 2019-09-17:
@tatokis AWESOMENESS :) Thanks a zillion :) works like a super charm..
[image]

closed 2019-09-17 by tatokis

cosmicchipsocket, 2019-09-18:
There are actually three RGB zones. The third one is in the front where the cable plugs in.

cosmicchipsocket, 2019-09-18:
I must have a different model or something. I have three extra programmable buttons on the side.

There are also two profile selection buttons, south of the mousewheel.

[image]

[image]

tatokis, 2019-09-18:
You have the wireless version of the mouse, which uses an entirely different protocol and won't be supported for some time.

cosmicchipsocket, 2019-09-18:
Is the protocol different even in wired mode?

It's not a big deal as I can program the hardware profiles via my laptop, though it is a bit inconvenient.

tatokis, 2019-09-18:
Yes. It's completely different and needs to be reverse engineered and implemented from scratch.

key60529, 2019-10-25:
hello there, I'm a Mac user which interested to buy the wireless version of the ironclaw, but after going through all the comments in this thread, it's seems not all the feature are support ATM except the rgb light control, am I right?

tatokis, 2019-10-25:
No. The wireless Ironclaw RGB is not supported. It will work as if there's no special software on the computer.

key60529, 2019-10-25:
> We can tell you what to do in IRC, where we'll be able to help you much faster.
> […](#)

Noted, I guess I will try my best to help after I get the mouse, will keep track on this thread and may reply soon once I have it.

paco3346, 2020-08-27:
Sounds like we still need to get a dump of the wireless protocol, right?

I'm a seasoned developer who just picked one of these up and am willing to pitch in with figuring out both the wired and wireless protocols.

tatokis, 2020-08-27:
@paco3346 if you have the ironclaw rgb, you can have a look at https://github.com/ckb-next/ckb-next/issues/558 and help around

I am currently working on getting the protocol documented and implementing basic support for it in the `bragi` branch.

commit d77d14a8 (2019-09-17): Merge branch 'ironclaw'

Closes #345