ckb-next/ckb-next #346: Harpoon Wireless RGB

ZirconiumX, 2019-01-09:
https://www.corsair.com/uk/en/Color/Region/HARPOON-RGB-WIRELESS-Gaming-Mouse/p/CH-9311011-EU

I suspect this will require wireless device support. Ugh.

ZirconiumX, 2019-01-18:
It also comes with a *new* new firmware format, as opposed to the *old* new firmware format of the Dark Core which used multiple files.

FlorianLaunay, 2019-02-06:
Harpoon Wireless RGB is also supplied with a cable. This mouse automatically switches to USB mode when connected to charge its battery.

Here is my **ckb-next-dev-detect** dump when this mouse is in USB mode :

```
ckb-next-dev-detect - Wed Feb  6 10:57:42 CET 2019

Kernel: Linux WKS-FLAUNAY 4.18.0-13-generic #14-Ubuntu SMP Wed Dec 5 09:04:24 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
LSB Version:	core-9.20170808ubuntu1-noarch:printing-9.20170808ubuntu1-noarch:security-9.20170808ubuntu1-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 18.10
Release:	18.10
Codename:	cosmic

USB Devices:

Bus 001 Device 005: ID 1b1c:1b5e Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b5e 
  bcdDevice            0.05
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR HARPOON RGB WIRELESS Gaming Mouse
  iSerial                 3 7AAB4F4FAF263270
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           91
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      60
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
[... 140 lines cut ...]

Daemon on last boot:
-- Logs begin at Mon 2018-09-10 09:19:11 CEST, end at Wed 2019-02-06 10:57:40 CET. --
Feb 05 11:56:09.904076 WKS-FLAUNAY systemd[1]: Started Corsair Keyboard Daemon.
Feb 05 11:56:09.908467 WKS-FLAUNAY ckb-next-daemon[940]:     ckb-next: Corsair RGB driver 0.3.2
Feb 05 11:56:09.908467 WKS-FLAUNAY ckb-next-daemon[940]: [I] Root controller ready at /dev/input/ckb0
Feb 05 18:13:33.162598 WKS-FLAUNAY ckb-next-daemon[940]: [I] Caught signal 15
Feb 05 18:13:33.162598 WKS-FLAUNAY ckb-next-daemon[940]: [I] Closing root controller
Feb 05 18:13:33.162598 WKS-FLAUNAY ckb-next-daemon[940]: [I] Removed device path /dev/input/ckb0
Feb 05 18:13:33.156224 WKS-FLAUNAY systemd[1]: Stopping Corsair Keyboard Daemon...
Feb 05 18:13:33.188112 WKS-FLAUNAY systemd[1]: Stopped Corsair Keyboard Daemon.

Dev nodes:
/dev/input/ckb0
/dev/input/ckb0/pid
15049
/dev/input/ckb0/version
0.3.2
/dev/input/ckb0/connected

Environment:

Kernel command line:
BOOT_IMAGE=/boot/vmlinuz-4.18.0-13-generic root=UUID=419757a9-64c2-4d26-9af1-4971f9ea5928 ro net.ifnames=0 biosdevname=0 quiet splash i915.enable_rc6=0 vt.handoff=1
```

ZirconiumX, 2019-02-06:
Ouch, that is very different to previous mice. A little bird told me that
the Harpoon Wireless has its own protocol though, so to get anywhere we'll
need packet traces and a lot of strong coffee.

FlorianLaunay, 2019-02-06:
So do you think you can add suport by the USB way ? This "new" protocol is also used in this mode ?

> Ouch, that is very different to previous mice. A little bird told me that the Harpoon Wireless has its own protocol though, so to get anywhere we'll need packet traces and a lot of strong coffee.

ZirconiumX, 2019-02-06:
At the moment we have no idea, but probably. However, given the Dark Core
has a (much) easier protocol that we still haven't finished due to burning
out on it, don't get your hopes up.

tatokis, 2019-02-06:
Can you run `sudo usbhid-dump -d 1b1c:1b5e` and post the output?

FlorianLaunay, 2019-02-06:
``` bash
[flaunay@WKS-FLAUNAY ~]$ sudo usbhid-dump -d 1b1c:1b5e
001:008:002:DESCRIPTOR         1549466160.761826
 06 42 FF 09 02 A1 01 15 00 26 FF 00 75 08 95 40
 09 02 81 02 C0

001:008:001:DESCRIPTOR         1549466160.762167
 06 42 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 09 01 81 02 95 40 09 01 91 02 C0

001:008:000:DESCRIPTOR         1549466160.762601
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 20
 15 00 25 01 95 20 75 01 81 02 75 10 95 02 05 01
 09 30 09 31 16 01 80 26 FF 7F 81 06 75 08 95 01
 05 01 09 38 15 81 25 7F 81 06 C0 C0
```

tatokis, 2019-02-06:
Can you compile this and run the following commands?
https://github.com/ckb-next/getUsbString

```
sudo ./getUsbString 0x1b1c 0x1b5e 4
```
and
```
sudo ./getUsbString 0x1b1c 0x1b5e 5
```

FlorianLaunay, 2019-02-07:
Sure. Here is the result :
```
[flaunay@WKS-FLAUNAY ~/getUsbString]$ sudo ./getUsbString 0x1b1c 0x1b5e 4
Read 54 bytes
Raw (hex): 36 03 42 00 6f 00 6f 00 74 00 6c 00 6f 00 61 00 64 00 65 00 72 00 20 00 76 00 65 00 72 00 73 00 69 00 6f 00 6e 00 3a 00 20 00 30 00 2e 00 35 00 2e 00 33 00 36 00 
String: Bootloader version: 0.5.36

[flaunay@WKS-FLAUNAY ~/getUsbString]$ sudo ./getUsbString 0x1b1c 0x1b5e 5
Read 10 bytes
Raw (hex): 0a 03 42 00 50 00 30 00 30 00 
String: BP00

```

alterxyz, 2019-04-20:
Any updates on this?

tatokis, 2019-04-20:
Unfortunately it will be a while, as it is a completely new protocol and there are already quite a few new devices that have been lined up for support for quite some time now.
Sorry for that

Trollwut, 2019-06-01:
So just to get a summary for now:

What *would* work with the new Harpoon device and what doesn't?

Currently thinking about getting one, as the USB/Bluetooth hybrid sounds very intriguing.

tatokis, 2019-06-01:
In theory it should just act as a normal mouse, without any software control (think of using it on windows without cue). In practice, I don't think anyone can say.

ThisNekoGuy, 2019-08-24:
Just got the mouse and it the compatibility notes say it's supported but I can't get CKB to recognize it even after following the command from the troubleshooting page...

tatokis, 2019-08-24:
The wireless one is explicitly not supported because it uses a new protocol. All devices in the list are wired unless otherwise specified.

luk-e, 2019-09-08:
Hi I have the wireless RGB Harpoon and ckb doesn't even see the device, it just cycles through all the colors like my keyboard (K68 RGB) does without ckb. Are there any fixes for this at the moment?

paulpopus, 2019-10-30:
Bumping this to show interest in CBK support.

For future readers:
- the mouse does work wireless with the dongle on Linux
- as far as i can tell it can remember your DPI but not RGB settings
- you can disable the annoying flashing of RGB if you connect it to an iCUE software on windows on wireless mode and enable power saving mode

rar0n, 2019-10-31:
@paulpopus 
Thanks for the suggestion!  Installed iCUE on a windows laptop just for that. I don't fancy the RGB flashing either, but a static color would be ok.

At least the battery probably lasts a bit longer this way. I just got this mouse btw, posting here to show interest in CKB support as well (I also have Corsair Strafe RGB mk 2 and using ckb-next, delayed start, Linux Mint 19).

Indeed it seems to remember the DPI color settings but not the logo color settings, when not connected to iCUE (odd!). Latest mouse firmware atm (v. 0.6.70).

gilvbp, 2020-04-12:
How could i help?

ZirconiumX, 2020-04-12:
You'll need to effectively reverse-engineer the Bragi protocol from Wireshark captures. This will be getting nowhere without it.

odelucca, 2020-07-17:
No updates on this?

andreldm, 2020-07-21:
> you can disable the annoying flashing of RGB if you connect it to an iCUE software on windows on wireless mode and enable power saving mode

I couldn't find "power saving mode" in iCUE, only an option to control brightness under settings. This is a good workaround for ppl that wants to get rid of this very annoying and battery waster flashing RGB. This preference even "survives" after iCue is closed.

I wasn't satisfied though and decided to give a try at reverse engineering this mouse's protocol, this is as far as I could get: https://gist.github.com/andreldm/367f0dba38da9c59e9f169fe0dd570fd

Downsides are: one needs a script to send a "keep alive" signal every 59 seconds and changing DPI clicking the central button won't work.

Hopefully this can useful to determine if this is a protocol already implemented in ckb-next or something entirely new. In any case, I'm available to further investigate how this protocol works and test ckb-next if anything promising gets merged.

toastyfe, 2020-10-21:
Hi everyone!

Any progress about this issue?

I just bought the mouse and I was hoping to turn off the RGB lights.

hockeymikey, 2020-10-21:
> Hi everyone!
> 
> Any progress about this issue?
> 
> I just bought the mouse and I was hoping to turn off the RGB lights.

Check out @andranoxi profile for now. He has a project on it and I"m using his python code myself in a service. Works fine, it will get you by until it's added in.

toastyfe, 2020-10-21:
Great! Unfortunately, I don't know how to proceed. Is there any step-by-step process I can follow?

hockeymikey, 2020-10-21:
> Great! Unfortunately, I don't know how to proceed. Is there any step-by-step process I can follow?

Sorry I meant @andreldm https://github.com/andreldm/harpoond

toastyfe, 2020-10-21:
Oh! Got it. But still, can't run `make` with it. It gives me:
```
No package 'libusb-1.0' found
gcc -O2 -Wall -Wextra  harpoond.c -o harpoond 
harpoond.c:1:10: fatal error: libusb.h: Arquivo ou diretório não encontrado
    1 | #include <libusb.h>
      |          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:7: harpoond] Erro 1
```

It seems that it is missing this `libusb-1.0` package.
And even so, how do I "tweak the harpoond.c" file to disable the RBG lights?

@andreldm can you help me out?

andreldm, 2020-10-22:
@toastyfe, yes, you need that package (`libusb-1.0-0-dev` in Ubuntu/Debian case). With regards to tweaking, you need to know a bit of hexadecimal to play with values in harpoond.c around line 115. If you have more questions please open an issue at [harpoond](https://github.com/andreldm/harpoond).

Lockna, 2020-11-29:
I have the Harpoon Wireless, with cable it works just fine but when I use the dongle it's "lagging". It just doesn't work smoth.
Is this is a problem with my mouse, or is this a generell problem?
Because I could help out reverse engineering and implement the wireless support

LevitatingBusinessMan, 2020-12-31:
If @andredlm found a way to set the colors and the dpi of the mouse, what's still needed for implementation in ckb-next and openrgb? How can I help?

andreldm, 2020-12-31:
> If @AndreDlm found a way to set the colors and the dpi of the mouse, what's still needed for implementation in ckb-next and openrgb? How can I help?

I think you wanted mention me. If you are interested, you can send a pull request to harpoond so I can review and merge it. With regards to ckb-next or openrgb you need to either ask/wait for maintainers to add support for this device or also send a pull request.

delucca, 2021-06-28:
No more updates on this? 😞

thompsonbrian, 2021-09-15:
The mouse "works" at the very least.  I'd like to adjust DPI settings straight from Linux (no access to a Windows machine), but I can manage with how it stands now.

catm4n, 2022-03-19:
was able to compile on arch linux but get the following error when attempting to load the daemon.

Device attached
short write (0, error -1)
Failed to release interface
Failed to claim interface
Failed to claim interface
Failed to claim interface
Failed to claim interface

valeth, 2022-04-04:
It does work as long as you don't want to turn off RGB and still keep your indicator.
Or change DPI settings.
Or use the DPI selector button, cause that does seem to attempt to use a protocol call that isn't yet implemented.

tatokis, 2022-04-04:
Patches welcome

RalphKemp, 2022-05-26:
> For what it's worth, there's been experimental support for the mouse in git for a while now. Hotplug support has been added as well, so it should be pretty usable. You just need to run ckb-next-daemon with `--enable-experimental`.

Thanks for this info - just got a Harpoon Wireless and the flashing colours were driving me a bit mad. Cheers!

quiquelhappy, 2022-06-07:
can someone make a comprenhensive guide on how I can make this work? I'm not familiarized with this repo

fritz-fritz, 2024-09-01:
Just thought I'd bump this for future readers.

- ckb-next works for Harpoon when launched with the `--enable-experimental` flag via USB or RF but not Bluetooth.
- To have a working (ish) DPI button rebind the "DPI Up" button to "Cycle Up"
- Logo and DPI color settings work along with animations
- Scroll wheel and forward/back buttons work by default but you can re-bind

The reason that I say DPI is working "ish" is because while I can cycle the indicator... I'm not convinced my DPI is actually changing.

To the devs: might be less confusing if the device image had accurate buttons for the mouse and the default mapping updated. I might take a look to see about submitting a pull request if I get the time but figured I'd share my experience here for others.

fritz-fritz, 2025-03-10:
Update:

- Use the patch from a2cc5cb33512f946eca1f6ee9bfcbe00576bbeb3 #909 to get the scroll wheel working.
- Use the patch from 4a1e9c85833eb29f580c32f32480ec5c9ce24c4d to get DPI actually working along with binding `DPI Up → DPI cycle up`

[image]

fritz-fritz, 2025-04-09:
@yaliv Yeah that patch is not perfect.

It also doesn't allow independent XY values. But without the patch it only changed the indicator so it's progress.

Thanks for the tip on the values! I hadn't determined why some seemed not to work.

ReemX, 2026-10-26:
Thanks @fritz-fritz for the write-up here, it got me all the way, and @gilvbp for 4a1e9c8.

On v0.6.2 + #1273 (the scroll fix) + 4a1e9c8, my Harpoon RGB Wireless (fw 5.0.129) works fully, wired and through the dongle, without experimental mode. The DPI button changes the real DPI, and logo and DPI lighting work separately (I keep the logo off for battery).

As far as I can tell, 4a1e9c8 isn't part of any PR and has diverged from master, so it could get lost. It would be worth turning it into a PR. The DPI button still needs the "DPI cycle up" rebind. I opened an issue for the root cause (the daemon reports this mouse as an Ironclaw Wireless): #1358