ckb-next/ckb-next #347: M65 Elite RGB

ZirconiumX, 2019-01-09:
https://www.corsair.com/uk/en/Categories/Products/Gaming-Mice/FPS-Fast-Action-Mice/M65-RGB-ELITE/p/CH-9309011-EU

This looks like the easiest of the three new releases to support, but I might well eat my words.

pkovacs, 2019-02-03:
~~~
# usbhid-dump -m 1b1c:1b5a
001:002:001:DESCRIPTOR         1549157482.740515
 06 C2 FF 09 04 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 B1 02 09 02 81 02 09 04 91 02 C0

001:002:000:DESCRIPTOR         1549157482.740838
 05 01 09 02 A1 01 85 01 09 01 A1 00 05 09 19 01
 29 20 15 00 25 01 95 20 75 01 81 02 05 01 09 30
 09 31 16 01 80 26 FF 7F 75 10 95 02 81 06 09 38
 15 81 25 7F 75 08 95 01 81 06 C0 C0 05 0C 09 01
 A1 01 85 02 05 0C 19 00 2A 3C 02 15 00 26 FF 0F
 75 10 95 02 81 00 C0 06 C1 FF 09 02 A1 01 85 03
 15 00 26 FF 00 95 3F 75 08 09 02 81 02 C0 06 C2
 FF 09 03 A1 01 85 0E 15 00 26 FF 00 95 3F 75 08
 09 03 81 02 C0 05 01 09 06 A1 01 85 05 05 07 19
 E0 29 E7 15 00 25 01 75 01 95 08 81 02 19 00 29
 97 15 00 25 01 75 01 95 98 81 02 C0
~~~

tatokis, 2019-02-03:
Try adding your mouse to the IS_MOUSE macro.
https://github.com/ckb-next/ckb-next/blob/master/src/daemon/usb.h#L169

ZirconiumX, 2019-02-03:
Could you also generate and attach a report for us?

pkovacs, 2019-02-03:
I'll try @tatokis IS_MOUSE first, gimme a minute.  @ZirconiumX, sure, just tell me what you require.

pkovacs, 2019-02-03:
Well the ckb-next software seems to be stable now and I have a new tab "CORSAIR M65 RGB ELITE".  The dpi selection seems to be working but the mouse colors are strobing around and I am not familiar enough with ckb-next or iCUE to know what's working and what isn't.  Progress, however!

pkovacs, 2019-02-03:
OK I found the 'Generate Report' button.  Attached.   There are some problems with the color selection.  The zones are not working correctly.   I am able to turn off the damn lights however and the dpi selection is working.   I notice that booting into Windows last time, the iCUE software stopped recognizing the mouse, as if ckb-next affected it somehow.   I had to reinstall iCUE to recognize the mouse again.  I hope that isn't an ongoing problem because I bounce back and forth a lot between the two OS'es.

[ckb-next-report-1549161542.txt](https://github.com/ckb-next/ckb-next/files/2825111/ckb-next-report-1549161542.txt)

pkovacs, 2019-02-03:
I added PR #366 as a my code reference.

ZirconiumX, 2019-02-03:
Running ckb-next should not affect CUE. Unless the M65 Elite uses file
hardware profiles, anyway.

The lights are probably messed up because they changed the numbering again.

To find them out we need to do a bit of manual intervention. First, make
sure the GUI is not running, just the daemon. Then, for each of [the 8
mouse RGB field names](
https://github.com/ckb-next/ckb-next/blob/master/src/daemon/keymap.c#L254)
(don't pay attention to what they're called; they match up to the M65 and
that's it), run `echo "active rgb $ZONE:ffffff" > /dev/ckb1/cmd` (assuming
your mouse is on the ckb1 node) and see if they do anything.

pkovacs, 2019-02-03:
My Vengeance K70 is on ckb1, the M65 on ckb2:
```
cd /dev/input/ckb2
echo active > cmd
echo active rgb wheel:ff0000 > cmd    # turns the wheel (top) zone red
echo active rgb dpi:ff0000 > cmd      # turns the dpi (middle) zone red
echo active rgb back:ff0000 > cmd     # turns the logo (lower) zone red
echo idle > cmd
```

No other labels cause any color change.  I think those three zones are the only light zones on the mouse.   The dpi0-5 colors and my changes to them were working I first got the gui running.

tatokis, 2019-02-03:
Also worthy of note, M65 RGB and the Pro use "front" for the wheel zone, whereas the Elite pro uses "wheel". Should fix that in the GUI

pkovacs, 2019-02-03:
So first I deleted all stock profiles and modes with ckb-next.   Then I created a profile called 'Phil' with one mode called 'Normal'.  I used the Lighting tab to disable all lights as best I could/  On the DPI tab I disabled the highest 2 dpis and set colors for the 3 I want to use.   Pressing the dpi button works as expected, allowing me to cycle through only 3 dpis.   I pressed 'Save to hardware'.   The profile 'Phil' and mode 'Normal' get a grey background splotch, I suppose that is your indicator for hardware-saved.  I killed ckb-next and the animation processes, leaving only the ckb-next-daemon.   The lights all came back on and the dpi colors are stock with operational 5 dpi buttons, telling me my 3 dpi profile is not active.  I entered the command mode of the daemon and did the following:

~~~
echo active >cmd
echo hwload >cmd
echo idle >cmd
~~~

The wheel and logo lights went out (conforming to my profile -- coincidence?) but the dpi lights and settings are back to the 5 stock settings.  Restarting ckb-next, I see my profile 'Phil' and mode 'Normal' are no longer grey-splotched and there is a new profile called 'Unnamed' with the grey splotch.  I deleted the Unnamed profile and switched back to my profile and all is well.

Ideally I would like not to have to run the ckb-next gui and the animation processes, instead just the daemon, but it seems I need to keep ckb-next active for now to keep my dpi settings correct and the lights off on the wheel and logo.  I hope this was helpful.

ZirconiumX, 2019-02-17:
FWIW, the "grey splotch" is actually a chip, i.e. the hardware profile. I wonder if the GUI tries to create a new profile if loading a profile fails. @tatokis?

tatokis, 2019-02-17:
I'm pretty sure it was fixed by https://github.com/ckb-next/ckb-next/commit/d5adfb15051c74104d90dac16704c4b008caa787 as others have reported the same issue with other mice.

That said, I'm definitely disabling hwload as it is clearly broken.

pkovacs, 2019-02-17:
I am trying master at commit 7cce023 and so far, so good with the M65 RGB ELITE mouse.  The Lighting section is responding properly to color changes.   I am also not seeing the creation of Unnamed profiles after rebooting a few times to test.  Good so far.

creatImre, 2019-06-13:
Hi, any progress with this @pkovacs ? This conversation is silent for months now. :(

tatokis, 2019-06-13:
There has been support for the M65 Elite RGB for some time now. If you're having trouble using it, make sure you are on the latest version, generate a report and create a new issue.

closed 2019-06-13 by tatokis

creatImre, 2019-06-13:
> There has been support for the M65 Elite RGB for some time now. If you're having trouble using it, make sure you are on the latest version, generate a report and create a new issue.

I missed to update to the latest version (completely forgot that apt upgrade not enough on ubuntu :) ), now it is working with the latest version. Thank you!!

pkovacs, 2019-06-13:
@creatImre  Yes, just run [release 0.4.0](https://github.com/ckb-next/ckb-next/releases/tag/v0.4.0) or later to see support for this mouse.  Cheers.

gravityfargo, 2024-05-02:
Note: This issue was closed because of #366