ckb-next/ckb-next #348: Question: How to do macros on OXS High Sierra?

tylersmith34, 2019-01-10:
# Something else

I have a k95 RGB and an MBP running High Sierra.  I'm trying to configure macros with the G keys and I am struggling with a way to do it.  I've read that the G keys don't know about key redirects, so mapping left control to left command, left windows to left option, and left alt to left control are unknown to the macros.  MacOS doesn't know about alt and windows keys, what do I do to make macros work?  The G keys are the reason I bought this keyboard (when I had a Windows laptop for work, now I was switched to Mac) and I'd hate to lose the G keys, they're very powerful when programming.

tylersmith34, 2019-01-10:
I tried the Google Group, no one has responded to my question.

ZirconiumX, 2019-01-10:
> Something else
>
> I have a k95 RGB and an MBP running High Sierra. I'm trying to configure
> macros with the G keys and I am struggling with a way to do it. I've read
> that the G keys don't know about [...]
This is correct; if you press left alt in a macro, you get left alt. While
it might be possible to change that in the daemon, the macro and rebinding
code is something we haven't touched.

> MacOS doesn't know about alt and windows keys, what do I do to make macros
> work?
>
No, but CKB knows about them, and the keys lalt and ralt are mapped to
option, and lwin and rwin are mapped to command.

> The G keys are the reason I bought this keyboard (when I had a Windows
> laptop for work, now I was switched to Mac) and I'd hate to lose the G
> keys, they're very powerful when programming.
>
Truthfully I haven't found I needed them with my Strafe RGB.

>

tylersmith34, 2019-01-11:
When I map the lwin, lalt, lctrl keys to mac keys, they work fine, it's just as macros they don't work. I've tried software like karabiner and using the Mac keyboard preferences to change bindings but nothing affects the G keys. 

I'm a heavy IntelliJ user and the G keys are a lifesaver, enabling me to be significantly faster.

ZirconiumX, 2019-01-11:
> When I map the lwin, lalt, lctrl keys to mac keys, they work fine,
>

They already *are* Mac keys.

>it's just as macros they don't work. I've tried software like karabiner

Karabiner is incompatible with CKB, so no, it won't work.

>and using the Mac keyboard preferences

The Mac keyboard preferences don't know what G-keys are, so no, it won't
work.

>to change bindings but nothing affects the G keys.

> I'm a heavy IntelliJ user and the G keys are a lifesaver, enabling me to
> be significantly faster.
>
I don't doubt that, but I'm confused what you're trying to do. What does
the macro record window say when you try to record, say, Command-C?

>

tylersmith34, 2019-01-11:
The macro record window shows the windows names, lwin, lalt, lctrl rather than the mac names, so the macros don't do anything even though I've mapped in CKB.  Without CKB the OS recognizes the control, windows, and alt keys as if they were on an Apple produced keyboard (that's what I meant by the OS keyboard preferences, not the G keys).  If I write lcmd in the macro, the key modifier is ignored. I can get screenshots tomorrow.

ZirconiumX, 2019-01-11:
CKB uses the Windows names, not the Mac names, because there's a Windows
key on your keyboard, not a command key.

tylersmith34, 2019-01-11:
Ok that makes sense.  Here is a screenshot of the macro for G16.  I also attached the ckb report.  When I use this G key, nothing happens.  I'm pressing the left CTRL key which is mapped to left command.
[image]

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2747802/ckb-next-dev-detect-report.gz)

[image]

ZirconiumX, 2019-01-11:
I'm pretty sure command is lalt not lctrl. Try that.

tylersmith34, 2019-01-11:
No, that creates ç.  I have this mapping:

| K95 key name (left side) | Mac key name |
| ---- | ---- |
Ctrl | Command
Windows | Control
Alt | Option

When I use `Record Macro` start button, `Ctrl` + `C` (K95 key names), it produces the screenshot from above `.+lctrl,+c,-lctrl,-c`

ZirconiumX, 2019-01-11:
Ignore the mapping. Since we've already established CKB ignores it (for
better or worse), it's irrelevant.

I guess the only alternative key would be Windows. Try Win-C.

ZirconiumX, 2019-01-11:
> That works! So I need to figure out what lalt and lctrl map to on the Mac
> keys and I can build the IntelliJ shortcuts but I won't be able to record
> them in CKB or on the keyboard.
>
> Thank you! You're restoring the joy of the K95!

Given what you said before, Windows control maps to Apple control, and
Windows alt maps to Apple option.

tylersmith34, 2019-01-11:
No not quite, Windows control maps to Mac Command, but you said that mapping is irrelevant but I think I see your point.  What threw me off is that `lwin` is making copy/paste work which is `command` + `c` on the Mac keyboard and that's not how I mapped `lwin`.  I thought I had tried `lwin` before, guess not.

ZirconiumX, 2019-01-11:
Yes, because the key corresponding to Windows and the key corresponding to
Command are the same key. If you plug a Windows keyboard into a Mac, and
press Windows, you will get Command. If you plug an Apple keyboard into a
PC and press Command, you will get Windows.

tylersmith34, 2019-01-11:
Right, I was starting to realizing that.  The mappings are really hard for me to ignore for some reason.  Thank you for shedding light on this for me!

closed 2019-01-11 by tylersmith34

ZirconiumX, 2019-01-11:
You're welcome.