ckb-next/ckb-next #350: Shortcuts not working properly in Chrome

KeizerDev, 2019-01-11:
Hi,

Since some weeks I got a strange problem when using Chrome. I can't actually use the shortcuts properly like I used before. Now you might think this is a chrome problem (and I guess so too) but there seems to be a fix when changing the keyboard language. I tried but i'm not sure how to do this for ckb-next. Does ckb-next just use the system information for keyboard layout or do I have to use the ckb-next tool?

I tried in ckb-next to set it to dvorak but it seems it doesn't changed anything on my layout.

More info about the issue is here:  https://www.reddit.com/r/chrome/comments/a764ww/help_shortcuts_stopped_working_on_newest_update/

Thanks in advance

ZirconiumX, 2019-01-11:
Can you send me your shortcuts?

tatokis, 2019-01-11:
This has nothing to do with ckb-next. To change the keyboard layout you have to use your DE or use `setxkbmap`. Ckb-next works at a lower level and doesn't have a concept of layouts. The setting in the user interface is purely visual.

KeizerDev, 2019-01-12:
@ZirconiumX some basic like ctrl+t is not working when i'm selecting a texarea.

@tatokis Thanks for your quick answer. I will close this issue then as it has nothing to do with ckb-next

closed 2019-01-12 by KeizerDev