ckb-next/ckb-next #352: H100i Pro support

skwerlman, 2019-01-12:
# New device support request

Detection report: [ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2751814/ckb-next-dev-detect-report.gz)

Let me know if i can provide any additional info to help get this supported

ZirconiumX, 2019-01-12:
This is supported by OpenCorsairLink: https://github.com/audiohacked/OpenCorsairLink/blob/testing/device.c#L295

Try that instead.

ZirconiumX, 2019-01-12:
Also you should *really* update your version of CKB-Next; 0.2.9 is pretty old.

skwerlman, 2019-01-12:
0.2.9 is the version being installed by nixos for some reason; i'm pretty sure its a config error on my end since the repos have 0.3.2; i'll def look at getting that sorted tho

i assume running ckb and ocl side-by-side is safe, ye?

ZirconiumX, 2019-01-12:
Yes, it's safe. We don't touch devices that OCL would use, and vice versa.