ckb-next/ckb-next #353: sync animations across multiple devices

xdotax, 2019-01-14:
# Feature request:
Sync profiles/modes/animations between devices. 

## What is the program's current behavior?
I have both the MM800 mousepad and K95P keyboard. I can set either one to a rainbow fade animation, but I can't sync the animation between devices. The best I can do is to start the animation on one device and then quickly start it on the other, but the animations will always be a half second apart. 

## What would you like it to do instead?
There are a few possibilities:
1. Have a button to restart the animations on all devices. That way, both devices will start at exactly the same time. 
2. Create a "combined device" that shows the RGB map of both devices on the same screen, and lets the user create modes that combine RGB lights from both devices
3. Create some other way to sync the same animation across multiple devices

## Any other notes & comments?
I searched the wiki, issues, and feature requests and didn't find anything about syncing between multiple devices. 

Thank you all for the work you are doing!

tatokis, 2019-01-14:
This has been on the todo list for a while, but unfortunately I have been too busy with other issues.
It probably won't be done soon, but I would love to see it at some point.

xdotax, 2019-01-14:
thank you for considering it! You all are doing great work on this project, keep it up. feel free to close this if appropriate.

tatokis, 2019-01-14:
Definitely no need to close it