ckb-next/ckb-next #359: Bug Installing on Ubuntu 14.04

Amarocq, 2019-01-20:
When I run the quick install I'm getting an error and a crash.  The problem seems to be with a member variable in the class QProcess that is referenced in animscript.cpp.   Thanks for taking a look.

## What is the program's current behavior?

[ 56%] Linking CXX static library libquazip.a
[ 56%] Built target quazip
Scanning dependencies of target ckb-next
[ 58%] Building CXX object src/gui/CMakeFiles/ckb-next.dir/animadddialog.cpp.o
[ 58%] Building CXX object src/gui/CMakeFiles/ckb-next.dir/animscript.cpp.o
/home/m_linux/ckb-next/src/gui/animscript.cpp: In member function ‘void AnimScript::readProcessErr()’:
/home/m_linux/ckb-next/src/gui/animscript.cpp:400:64: error: ‘class QProcess’ has no member named ‘processId’
     QString out = _info.name + " (" + QString::number(process->processId()) + ")";  //************************************
                                                                ^
make[2]: *** [src/gui/CMakeFiles/ckb-next.dir/animscript.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [src/gui/CMakeFiles/ckb-next.dir/all] Error 2
make: *** [all] Error 2
An error occurred,
press enter to exit.

tatokis, 2019-01-20:
Can you run `qmake --version` ?

Amarocq, 2019-01-20:
Sure, this is what I got:

QMake version 3.0
Using Qt version 5.2.1 in /usr/lib/x86_64-linux-gnu

tatokis, 2019-01-20:
Should be fixed in master by https://github.com/ckb-next/ckb-next/commit/0932a819c5539f4eefb3dc1d33920d605a80b0df

Amarocq, 2019-01-20:
It installed successfully after the update.  Thanks again for the help.