ckb-next/ckb-next #361: mviz: sound input get's laggy and stale processes are left behind if volume changed

WildPenquin, 2019-01-23:
mviz seems to work fine, but as soon as I change the volume, it get's laggy (i.e. the visualization is way behind, up to tens of seconds or minutes). Also, I noticed mviz processes are left behind even if I change the mode to one that does not use audio input. Ckb keeps spamming these processes. It **might** be that these processes are not spammed if I have not changed volume.

## General information

- Built from ckb-next-git package in AUR yesterday.
- Keyboard: Corsair K95 RGB Platinum
- Running Pulseaudio, sound card is _HDA Intel PCH, ALC1150 Digital_: there is also DisplayPort output (which I don't mostly use), no other audio devices (currently).
- At least playing Youtube streams in Firefox and music via Clementine can reproduce this issue (didn't try other sources yet, but I would be surprised if that makes a difference!)

## What is the expected behavior of the program?
- Changing volume should not cause a latency issue
- mviz -processes should exit gracefully if not in use anymore. At least spamming unnecessary processes should stop.

WildPenquin, 2019-01-23:
Noticed that if I cycle **slowly** trough modes that use music visualization, the stale processes are removed. It is certainly possible there will be only **one** left-over process per mode, which gets re-created / deleted once the mode is re-cycled trough.

tatokis, 2019-03-05:
So, I've been noticing this as well, and I am pretty sure it's a pulseaudio/libpulse bug.
you can try something like `PULSE_LATENCY_MSEC=30 ckb-next` and see if it helps

closed 2022-10-03 by tatokis