ckb-next/ckb-next #364: [OSX-Mojave] [K95 Platinum RGB] - Save to Hardware disabled, Layout not load, Binding

hhl88, 2019-01-29:
Hi there,
I have created a profile for my keyboard, and cannot use "Save to Hardware" function. (sometimes after restarting my profile disappears)

Also after restarting or reopening app again, Layout always change to default English, although i have chosen German layout.

I tried to Binding function, such as Keyboard like Function(think that is FN), or Program("Lauch program on key press" - /Applications/Safari.app) but nothing works.
Thanks a lot

ZirconiumX, 2019-01-30:
> I have created a profile for my keyboard, and cannot use "Save to Hardware" function. (sometimes after restarting my profile disappears)

We have no support for K95 Platinum hardware profiles, so it's greyed out.

> Also after restarting or reopening app again, Layout always change to default English, although i have chosen German layout.

@tatokis - we save locale, right?

> I tried to Binding function, such as Keyboard like Function(think that is FN), 

Do you mean the Function found on Apple keyboards?

> or Program("Lauch program on key press" - /Applications/Safari.app) but nothing works.

As I understand it, that is not a program, actually. Mac OS X pretends it's a program to the interface, but it's actually a folder. Launching a folder doesn't work.

tatokis, 2019-01-31:
@ZirconiumX we do, however the issue is a bit complicated. I thought I had fixed it though.

@hhl88 Are you running version 0.3.2?

And as @ZirconiumX said, you need to put "open /Applications/Safari.app"
We really should add a file browser asap to that.