ckb-next/ckb-next #365: [Bug Report] Forgets key bindings and modifier key preferences

SSmale, 2019-01-31:
# Bug report

## General information
Hi, I am using this to run my Scimitar mouse and it is amazing. I recently got the K70R keyboard and I am having some issues. 

I have changed the modifier key setting to match the standard Mac layout (ctrl,opt,cmd) and the print screen key to Eject to allow me to put he Mac to sleep.

During use and after sleep the modifier keys will randomly change back to the default (ctrl,cmd,opt) and the eject key working is sporadic at best. 

### Source
  * pre-built pkg for macOS - ckb-next 0.3.2

## Logs & crash reports
[ckb-next-report-1548921454.gz](https://github.com/ckb-next/ckb-next/files/2816173/ckb-next-report-1548921454.gz)

## What is the program's current behavior?

Sometimes map keys as per settings, sometimes not. key rebinding rarely works. 

## What is the expected behavior of the program?

To consistently rebind the keys and modifiers. 

## Any other notes & comments?

tatokis, 2019-01-31:
I think this might be related to https://github.com/ckb-next/ckb-next/issues/324#issuecomment-459447961

I have just made a build for that, so can you also please try the build from that post?

SSmale, 2019-01-31:
Hi @tatokis,

I have used that build and it looked good until I put the Mac to sleep and it reverted back to the defaults as described above

Simon

SSmale, 2019-09-06:
I have now changed keyboard to the K70 RGB and this seems to be fixed, might have been an issue with the keyboard itself

closed 2019-09-06 by SSmale