ckb-next/ckb-next #368: GLAIVE doesn't work after sleep on Ubuntu 18.04

dominalien, 2019-02-06:
# Bug report

## General information

First of all, thank you very much for the awesome ckb-next.

I just built ckb-next from git and now Glaive mouse doesn't work after suspend. The keyboard (Strafe) works. This used to happen very rarely with the version I had installed (compiled from git a couple of months ago), but now it happens every time.

### Source
built manually using quickinstall script

## Logs & crash reports
I attach ckb's report. I don't know what info it contains, but this is generated after sleep when the mouse wasn't working and after replugging it so it now works.

## What is the program's current behavior?
Glaive mouse doesn't work after waking from suspend

## What is the expected behavior of the program?
Glaive mouse should work after waking from suspend

I'll be happy to provide any other info/logs and test new versions. Thanks!

dominalien, 2019-02-06:
[ckb-next-report-1549485278.gz](https://github.com/ckb-next/ckb-next/files/2838150/ckb-next-report-1549485278.gz)

apathyrecharge, 2019-02-13:
hope i'm not hijacking your issue here, but i seem to be having a similar issue with a K70 & M800 Polaris on arch. when i wake my machine from sleep the keyboard is completely unresponsive - no rgb animations and no input is registered. the polaris mousepad can still power my logitech mouse but the rgb animation on the mousepad changes to the stock rainbow pattern.
from here i just try reboot the machine but it will hang at "kvm: exiting hardware virtualization".

this happens whether i install through the AUR or compile the master git (although i think AUR just fetches latest commit anyway)
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2862370/ckb-next-dev-detect-report.gz)

tatokis, 2019-02-13:
Can you try the code from the fix-sleep-input-thread branch? 
https://github.com/ckb-next/ckb-next/compare/fix-sleep-input-thread

apathyrecharge, 2019-02-13:
no joy, the issue persists. [heres another report](https://github.com/ckb-next/ckb-next/files/2862454/ckb-next-dev-detect-report.gz) if its useful

ZirconiumX, 2019-02-14:
Here's my theory.

Corsair devices have a hardware and software mode. They start in hardware
mode, which sends HID input packets beginning with 01, and then we switch
it to software mode, which sends the same 01 packets but also
Corsair-specific 02 and 03 packets. When in software mode, we discard (most
of) the 01 packets and listen for the 02/03 packets.

Linux doesn't tell us when suspend is happening; it's (supposedly)
transparent to the application, *but* it also powers off the keyboard.

When you resume from suspend, the keyboard and mousepad start in hardware
mode, but ckb-next continues operating in software mode.

For the K70, we could detect receiving 01 but not 02/03 packets and guess
that we resumed from suspend, but we have no way of telling that if, for
example, somebody has only a Polaris, because the Polaris does not have any
input packets.

It's a tricky problem, and perhaps the only reliable method is having a GUI
button that says "Reinitialise All" or something like that.

dominalien, 2019-02-14:
So now it turns out the issue is different and much wider than I'd initially thought. My keyboard (Strafe RGB) is now acting up as well. As an example: yesterday, I booted the computer and then put it to sleep on the login screen. After waking it up, both the mouse and keyboard were working. I then logged in and as ckb-next was loading they both stopped working. Replugging both fixed the issue.

Just to reiterate, this issue is new. I'd been using ckb-next for months without problems. It did happen very sporadically that the mouse wouldn't work after sleep, but it was basically a non-issue. Now, since I updated ckb-next last week, it happens all the time. And the keyboard started doing the same thing, to boot.

I have not tried yet the branch tatokis suggests, I will when I get home.

I am more than willing to provide additional information to help get this fixed, just tell me what to send.

dominalien, 2019-02-14:
A button to reinitialize all would be rather useless if the kb/mouse aren't working.

Isn't there a way to run something like a script on sleep and resume? Perhaps ckb-next could hook into that.

ZirconiumX, 2019-02-14:
The problem is that sleep and resume is entirely transparent to us; we have
*no way* of knowing when we woke from suspend.

dominalien, 2019-02-14:
Yeah, I get that, but if I understand correctly it's possible in Linux to run a script or program after it wakes up from sleep and that could be used to inform ckb-next that it needs to reinitialize devices?

tatokis, 2019-02-14:
As far as I'm aware it's systemd only. 

That said, can you try reverting https://github.com/ckb-next/ckb-next/commit/106c474462a5aad166144128604616800b57e365

dominalien, 2019-03-05:
It seems this is now fixed. I can sleep again, yay. :-)

closed 2019-03-05 by dominalien