ckb-next/ckb-next #371: [OSX 10.14.1][Scimitar] Mouse4/5 not working

asteed, 2019-02-08:
# Bug report

## General information

### Source
  * pre-built pkg for macOS

## Logs & crash reports

Attach:
 * The generated report from the section above 
[ckb-next-report-1549654022.gz](https://github.com/ckb-next/ckb-next/files/2846579/ckb-next-report-1549654022.gz)

## What is the program's current behavior?
 * Binding a thumb button (1) to `Mouse > Special Button > 4 (Back)` performs a Mouse 3 (Middle Mouse) Scroll on keydown, and cancels it on keyup.

## What is the expected behavior of the program?
 * Binding a thumb button (1) to `Mouse > Special Button > 4 (Back)` performs a Mouse 4 (Back) event.

[image]

## Any other notes & comments?
Other bindings, such as binding a thumb button (1) to `Keyboard > Typing > Q` inserts a `Q` on press as expected.

ZirconiumX, 2019-02-08:
This is an unfortunate consequence of macOS not having support for backwards/forwards. The keys are I believe default bound to a macro that presses Cmd-[ and Cmd-] which seem to be the standard shortcuts for this, but they're not perfect.

asteed, 2019-02-08:
Ok, understood. I'm unable to get the macro feature to record keypresses in this case. 
* Enter macro back `Back`
* Click Start
* Press `Cmd+[`
* Click Stop
Only `+mouse1,-mouse1` is shown in the actions box, assuming its only capturing when I click the start/stop buttons.. 
[image]

ZirconiumX, 2019-02-08:
ckb-next only captures events from the device you set the macro on due to
internal issues. You'll need to type in the macro -
"+lalt,+lbrace,-lbrace,-lalt" from memory - manually.

davidbdeath, 2019-02-10:
it's "+lwin,+lbrace,-lbrace,-lwin". Corsair released a version of iCUE for mac but.... You can't even map shift to a key on the scimitar. They will probably have some lame excuse as to why.

asteed, 2019-02-21:
Thanks @davidbdeath - that worked. I assume [this is a decent list](https://github.com/ckb-next/ckb-next/blob/30ebedf30312c3b2ce31d815bd9ae5c46e077848/src/daemon/keymap.c) of possible keynames that can be used

macakanonimni, 2019-10-05:
Hi!
I am having the same problem. I have a Corsair Sabre RGB mouse which has the foward/back mouse button and again when I bind something to it, it does nothing. My goal is to bind shift on the back and ctrl on the forwards button, but I have no clue how to fix it, neither I understand what should I be doing (setting macro or programs on the ckb site). I checked out similar topics, but am not sure what should I do to set the mouse properly. I tried setting it up on the Binding tab of the mouse button but it does nothing. Any help would be really appreciated.

closed 2022-03-29 by tatokis