ckb-next/ckb-next #373: Corsair K68 stopped working after installing ckb-next

DylanM923, 2019-02-16:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.

This issue template has 3 main sections of which you should use only 1:
- [New Device Support Request](#new-device-support-request),
- [Feature Request](#feature-request), or
- [Bug Report](#bug-report).

If your issue doesn't fit in any of those categories, please use the [#something-else] section at the bottom.
Delete the redundant sections before sending.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

---

# New device support request

Coersair K68 RGB.  [New Device Support Request](#new-device-support-request),

---

# Feature request
Working keys, preferably with customized RGB.

## What is the program's current behavior?
Will not allow me to type anything after installing the cab-next software. I can customize my RGB and based off of the ripple effect, it registers me pressing the keys, but I cannot type anything. Even after uninstalling the software and restarting my computer. I saw something about trying to flip a pole switch but I don't believe my keyboard has one.

## What would you like it to do instead?
Allow me to type.

## Any other notes & comments?
I used the pre made MacOS download for ckb-next. I would like to say that before I installed the software, everything was working perfectly fine, the keyboard is still relatively new and I have never had any problems with it before. It's really frustrating to have it suddenly stop working after installing this, all I wanted was more customization.
---

tatokis, 2019-02-16:
Please click the generate report button in the settings tab of the ckb-next app and attach the report. 

You can run this command in a terminal, and then unplug/replug your keyboard to temporarily disable ckb-next until you reboot. 

sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist

n266hz, 2019-02-20:
I have the exact same issue with my k70 rgb, kind of
When ckb is running i can change RGB setting but cant type anything, keyboard isnt working at all
If I unplug/replug my keyboard, it works but program says "no device connected", then a restart my macbook and same story happens again
macos 10.14

tatokis, 2019-02-20:
> Please click the generate report button in the settings tab of the ckb-next app and attach the report.

alucryd, 2019-03-01:
[ckb-next-report-1551422458.gz](https://github.com/ckb-next/ckb-next/files/2918247/ckb-next-report-1551422458.gz)

Same thing with a regular K70, been going on for as long as I can remember. Attached a report. Is that something that something that is covered by usbhid and usbcore quirks?

DylanM923, 2019-03-01:
So I fixed my K68. Turns out I made a stupid mistake and I had to allow
something else, alongside the normal CKB to make changes to my computer and
then my keyboard started working again. I don’t have my MacBook with me
currently so I can’t give you exact details, but just go to system
preferences and in the security section go to the far right I believe, and
make sure you have everything checked to make changes that you need

alucryd, 2019-03-01:
Thanks, I'm not on MacOS, but Arch Linux though. But I have the same symptoms.

tatokis, 2019-03-01:
@alucryd The symptoms definitely look like what would be fixed by the usbcore quirks. They don't show up in your cmdline, so if you tried to, you haven't added them properly.

Do report back if the usbcore quirks fix your issue

alucryd, 2019-03-01:
Thanks for the feedback, I haven't tried them yet, but will as soon as possible.

alucryd, 2019-03-03:
Looks like that did the trick, thanks a lot!

closed 2019-03-05 by tatokis