ckb-next/ckb-next #374: ckb-next-daemon regression. Keyboard timing out/failing to reset.

Cloaked9000, 2019-02-16:
# Bug report

## General information
There seems to have been a recent regression in the ckb-next daemon, with my Corsair Gaming K70 LUX RGB Keyboard. Up until today I've been running 8db1a7d112d763660599764be0f351f96061a571 with no issue. I decided to update today, and downloaded the latest source. Upon reboot, the keyboard didn't show up on the GUI, and I couldn't type with it. Periodically, the keyboard would flash up on the GUI for half a second, and the keyboard would light up, before disappearing again. I created a report from the GUI (see attached), and downgraded to the previous version that I was using (which works fine).

### Source
Built from the latest source, commit cf0d31f3ea9553d9f14c84a531291299a1f82b01.

## Logs & crash reports
Generated report: 
[ckb-next-report-1550333584.gz](https://github.com/ckb-next/ckb-next/files/2872175/ckb-next-report-1550333584.gz)

tatokis, 2019-02-17:
Try the usbcore quirks from here https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbcore-quirks

Cloaked9000, 2019-02-18:
That does appear to have worked thanks ^^

Is it a bug that the workaround is only needed in newer versions of ckb-next, or is the behaviour correct, if you don't mind me asking?