ckb-next/ckb-next #378: SysRq key not working on K95 RGB Platinum

GoldsteinE, 2019-03-05:
# Bug report

## General information
K95 has no SysRq key, but Print Screen also doesn't work as SysRq in combinations like Alt+SysRq+H or Ctrl+Alt+SysRq+H.
### Source
  * ckb-next package from AUR

## Logs & crash reports
[ckb-next-report-1551782588.gz](https://github.com/ckb-next/ckb-next/files/2930569/ckb-next-report-1551782588.gz)

## What is the program's current behavior?
Magic SysRq keystrokes just don't work

## What is the expected behavior of the program?
PrintScreen should work as SysRq

## Any other notes & comments?
I'm not sure that getting SysRq key working is possible on ckb-next level, but config window shows me "Print Screen" as "Print Screen/SysRq" and that gives me hope

ZirconiumX, 2019-03-05:
Print Screen and SysRq are the same keycode.

Honestly I'm wondering if you actually have sysrq enabled; try checking the value of `sysctl kernel.sysrq` first.

tatokis, 2019-03-05:
It definitely works for me.
```
[231021.304870] sysrq: SysRq : HELP : loglevel(0-9) reboot(b) crash(c) terminate-all-tasks(e) memory-full-oom-kill(f) kill-all-tasks(i) thaw-filesystems(j) sak(k) show-backtrace-all-active-cpus(l) show-memory-usage(m) nice-all-RT-tasks(n) poweroff(o) show-registers(p) show-all-timers(q) unraw(r) sync(s) show-task-states(t) unmount(u) force-fb(V) show-blocked-tasks(w) dump-ftrace-buffer(z) 
```
That said, if userspace freezes or crashes, SysRq will be useless as the daemon won't be able to process messages.
I usually have a dedicated PS/2 keyboard for dealing with those.

GoldsteinE, 2019-03-05:
```
$ sysctl kernel.sysrq
kernel.sysrq = 1
```

WildPenquin, 2019-07-04:
SysRq definitely works.

Original reporter is using Arch. It seems Arch has SysRq disabled per default. It needs to be enabled as outlined here: https://wiki.archlinux.org/index.php/Keyboard_shortcuts . I can confirm SysRq does work on K95 RGB Platinum (in Arch Linux after enabling it).

So, not K95 bug. I have hit a "problem" just a few times because of a program in a freak runaway memory leak, consequently userspace in a near-freeze, including ckb-daemon; I was **not** being able to do anything with the keyboard reliably. I couldn't type in my password (to kill the process eating RAM with sudo) since the keyboard was too laggy (and I could not see the password as I type it, to know if it will register correctly). As I understand, there is little that could be done for this issue? Unless there is way to "force" a process to stay in RAM? However, that is a whole another "issue". FWIW SysRq reactions were still getting trough, albeit slowly (after 30-60 seconds).

I think this should be closed as not ckb-next bug?

closed 2019-08-16 by tatokis