ckb-next/ckb-next #380: Wiki Update

hevanaa, 2019-03-06:
On wiki page https://github.com/ckb-next/ckb-next/wiki/Linux-Installation

ckb-next is now available in stable Fedora 28 and 29 repositories:

https://bugzilla.redhat.com/show_bug.cgi?id=1670508

I'm no longer the maintainer of the package in Fedora. You could update the wiki and remove my personal johanh/ckb repo (I won't probably update it anymore). If someone else want to maintain the CentOS/RHEL build they can copy it freely.

hevanaa, 2019-03-06:
You can now install the package from the official Fedora repo with 'sudo dnf install ckb-next'. I guess it will also soon become available in Gnome Software in Fedora (when appstream data is updated).

ZirconiumX, 2019-03-17:
I just updated the wiki page, hopefully it looks okay.

closed 2019-03-17 by ZirconiumX