ckb-next/ckb-next #381: Music Visualizer Altered After Update

RPINerd, 2019-03-08:
## General information

CKB-Next 0.3.2-102-g6fdf6c4
K70 LUX RGB

### Source

Built manually using the "install_cbk.sh" script. Contents are as follows:

`echo "Installing Dependencies"
sudo pacman -S --needed base-devel cmake libsystemd qt5-base zlib libappindicator-gtk2 quazip libpulse git
echo "Cloning CKB-Next.git"
cd /home/chris/Scratch/
git clone https://github.com/ckb-next/ckb-next.git
cd ckb-next
echo "Running Install"
./quickinstall`

## Logs & crash reports

[ckb-next-report-1552017018.txt](https://github.com/ckb-next/ckb-next/files/2944322/ckb-next-report-1552017018.txt)
Here is the generated report from the settings section of CKB-Next, but crashing is not the issue.

## What is the program's current behavior?

With the "music visualization" animation enabled, every key of the keyboard glows with the selected color in unison.

## What is the expected behavior of the program?

Prior to updating just now, the colors would appear as an equalizer, with the keys lighting up only when that select frequency reached a certain volume. As an example; before updating if I had a low bass tone thumping, color would jump from Left-Alt and Space Bar filling in to 'E' '3', and if the volume was loud enough, some of the lower function keys like 'F2' and 'F3'. Meanwhile, all other lighting on the board would remain at the default. Post update, if I play that same thumping bass tone, EVERY key on the keyboard lights with intensity proportional to the volume. 

## Any other notes & comments?

I know the first question will probably be "what version did you upgrade from?" and I really wish I could answer that! Short of doing a timeshift restore, I'm not sure how I would go about figuring out what I upgraded from. If there is an easy (non-destructive) way to do this let me know and I'll get back to you ASAP!

tatokis, 2019-03-08:
I rewrote the music visualiser, as the old one has various problems. In the process, I added a single colour mode which has the effect you describe. It is supposed to be disabled by default, but due to a bug, it was enabled by default.

You can either readd the visualiser in your profile, deleting the old one, or go to "edit properties" and uncheck `single colour`.

I have fixed this issue in commit https://github.com/ckb-next/ckb-next/commit/d5e080b1607f4a1b0cd00464d0519a0259a8f431

closed 2019-03-08 by tatokis

RPINerd, 2019-03-08:
Awesome thank you! This is exactly what I was looking for!

LinuxGameGeek, 2019-04-12:
I cannot change the input device for the music-visualizer. There is no field to enter a number in the music visualization settings window as it is shown in https://github.com/ckb-next/ckb-next/wiki/Animations#mviz-music-visualizer. Help!

tatokis, 2019-04-12:
All you need to do is install and run pulseaudio volume control `pavucontrol` while the music visualiser is running. You can then switch to the recording tab and change the device there, as shown on the first part here https://github.com/ckb-next/ckb-next/wiki/Animations#on-linux-with-pulseaudio

LinuxGameGeek, 2019-04-12:
I already installed it. I only can choose from microphones and not speaker output.

LinuxGameGeek, 2019-04-12:
Thanks now it works.;)