ckb-next/ckb-next #385: Layout of keyboard is not saved

lukasbash, 2019-03-14:
# Bug report

## General information
Using Arch Linux 4.20.13 either with GNOME as desktop environment or awesomewm as a window manager.
Saving the device's layout does not work for me.
I start up my machine and as soon as the daemon has been started up, I set the layout to e.g. German, then restart or shut down the machine and the setting is gone.

### Source
Installed via yay from AUR.
https://aur.archlinux.org/packages/ckb-next/

## Logs & crash reports
[ckb-next-report-1552546987.gz](https://github.com/ckb-next/ckb-next/files/2965067/ckb-next-report-1552546987.gz)

## What is the program's current behavior?
It does not save the keyboard layout.

## What is the expected behavior of the program?
It should save the keyboard layout and persist it also over a restart or shutdown.

## Any other notes & comments?
No.

ZirconiumX, 2019-03-17:
Layout detection involves a bit of guesswork on our end. We have no way of knowing what the keyboard is, apart from physical layout. So we just pick ISO English and hope for the best. I'm sorry, but it's the best we can do at the moment, unless we go to the trouble of trying to work out what the virtual keyboard layout is meant to be.

lukasbash, 2019-03-18:
@ZirconiumX 
Thank you for your reply!
So far I was not talking about the actual layout detection, but rather saving the setting that I set manually in the GUI to a configuration that is loaded during startup.
I assume at some point you have to write stuff into a device specific configuration (e.g. animations, etc.) so why not store the layout as well?

ZirconiumX, 2019-03-18:
We do save the physical layout, as far as I know, but not the virtual
layout, because if a Brit imported a French profile and started typing in
AZERTY, they would be understandably quite confused.

Either way, @tatokis handles GUI things, so I'm going to let him reply.

tatokis, 2019-03-18:
There are two bugs here.
One is
```
Mar 13 06:44:02.001291 jupiter ckb-next-daemon[690]: [W] os_usbrecv (via firmware.c:24): Interrupt cond error 110
Mar 13 06:44:02.001291 jupiter ckb-next-daemon[690]: [W] _start_dev (device.c:43): Unable to load firmware version/poll rate
```

And the other one is because of that, the GUI fails to save the preference. I thought I had fixed it, but apparently not.

lukasbash, 2019-03-19:
> And the other one is because of that, the GUI fails to save the preference. I thought I had fixed it, but apparently not.

I assume a fix cannot be estimated right now? :yum:

tatokis, 2019-03-19:
Since you're on linux, I think it's more important to try to figure out why the keyboard isn't responding to our packets. If you wish, you can drop by on [IRC](http://webchat.freenode.net/?channels=%23ckb-next&uio=d4) and we can try to figure out why that part is not working.

carlmlane, 2019-06-14:
I can recreate this on macOS Mojave 10.14.5 (18F132) with Corsair K70R

Is there anything else I can provide to help this issue along?

tatokis, 2019-06-14:
This has been fixed in https://github.com/ckb-next/ckb-next/commit/11af88ddff7e07a76c4f9a7590959b04779f5e6b 
The fix should be included in the next release, which we should do relatively soon.

closed 2022-10-03 by tatokis