ckb-next/ckb-next #386: M65 RGB Not Being detected

PenguinOJ, 2019-03-14:
#bug-report

This issue template has 3 main sections of which you should use only 1:
#bug-report

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

ckb-next-dev-detect - Thu Mar 14 15:47:38 CDT 2019

Kernel: Darwin Joses-iMac.local 18.2.0 Darwin Kernel Version 18.2.0: Thu Dec 20 20:46:53 PST 2018; root:xnu-4903.241.1~1/RELEASE_X86_64 x86_64

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.14.3
BuildVersion:	18D109

USB Devices:
        Corsair Gaming M65 Pro RGB Mouse:

          Product ID: 0x1b2e
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 2.06
          Serial Number: 0B02C039AECB9CA8577A81B5F5001940
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14400000 / 12
          Current Available (mA): 500
          Current Required (mA): 300
          Extra Operating Current (mA): 0

IOReg:
+-o Corsair Gaming M65 Pro RGB Mouse@14400000  <class AppleUSBDevice, id 0x100000857, registered, matched, active, busy 0 (2 ms), retain 18>
  | {
  |   "sessionID" = 0x1f7fb6e551f
  |   "iManufacturer" = 0x1
  |   "bNumConfigurations" = 0x1
  |   "idProduct" = 0x1b2e
  |   "bcdDevice" = 0x206
  |   "Bus Power Available" = 0xfa
  |   "USB Address" = 0xc
  |   "bMaxPacketSize0" = 0x40
  |   "iProduct" = 0x2
  |   "iSerialNumber" = 0x3
  |   "bDeviceClass" = 0x0
  |   "Built-In" = No
  |   "locationID" = 0x14400000
  |   "bDeviceSubClass" = 0x0
  |   "bcdUSB" = 0x200
  |   "USB Product Name" = "Corsair Gaming M65 Pro RGB Mouse"
  |   "PortNum" = 0x4
  |   "non-removable" = "no"
  |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |   "bDeviceProtocol" = 0x0
  |   "IOUserClientClass" = "IOUSBDeviceUserClientV2"
  |   "IOPowerManagement" = {"DevicePowerState"=0x0,"CurrentPowerState"=0x3,"CapabilityFlags"=0x10000,"MaxPowerState"=0x4,"DriverPowerState"=0x3}
  |   "kUSBCurrentConfiguration" = 0x1
  |   "Device Speed" = 0x1
  |   "USB Vendor Name" = "Corsair"
  |   "idVendor" = 0x1b1c
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "USB Serial Number" = "0B02C039AECB9CA8577A81B5F5001940"
  |   "IOClassNameOverride" = "IOUSBDevice"
  | }
  | 
  +-o Google Chrome  <class IOUSBDeviceUserClientV2, id 0x10000085b, !registered, !matched, active, busy 0, retain 5>
  |   {
  |     "IOUserClientCreator" = "pid 273, Google Chrome"
  |   }
  |   
  +-o AppleUSBInterface@0  <class AppleUSBInterface, id 0x100000870, registered, matched, active, busy 0 (0 ms), retain 5>
[... 160 lines cut ...]

(Sorry couldn't figure out to just put the file)

# Bug report

## General information

### Source
 PreBuilt for MacOS

## Logs & crash reports

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behavior?
Just wont detect my mouse
## What is the expected behavior of the program?
To detect my mouse?
## Any other notes & comments?

---

# Something else

PenguinOJ, 2019-03-14:
I now fixed the orginal problem but now it won't Right Click, Left click, etc... help?

ZirconiumX, 2019-03-17:
I don't think you have the Karabiner kext installed/enabled properly, but it's difficult to tell because you copy/pasted the report instead of uploading the whole file, making it difficult to read.

tatokis, 2019-03-18:
> [F] ckb-next-daemon is already running (PID 571). Try killall ckb-next-daemon.
> [F] (If you're certain the process is dead, delete /var/run/ckb0/pid and try again)

Reboot and generate another report

closed 2022-10-03 by tatokis