ckb-next/ckb-next #388: Financial Contribution

guyewhite, 2019-03-17:
# Something else

I want to financially contribute to this project. It has saved me considerable time and expense and I'd like to reciprocate. Is there some way I can financially support this project? Who is the project lead?

ZirconiumX, 2019-03-17:
This project is maintained by two people, @tatokis and myself.

I need money to eat at university, but I worry that if people throw money at me for this, they'll expect major progress on this.

Progress on ckb-next has slowed not due to time or money, though more of both would always be welcome. It has slowed due to both developers burning out (which is why #158 has stalled; it needs refactoring and I don't feel like it), and due to legal concerns over binary blobs (which is why #136 has stalled; we believe Corsair send a binary blob of ARM code to the CPU, and we really don't want to annoy them).

If somebody donates money to me, I want them to promise that it is a) easily affordable for them, and b) given without expecting anything in return.

ZirconiumX, 2019-03-17:
That said, my PayPal email is dan.ravensloft@gmail.com.

tatokis, 2019-03-19:
> Who is the project lead?

That'd be me.
If you wish, you may contact me via email on the address listed on my github profile.

Either way, since this is not an issue, I'll close it. Thank you nonetheless.

closed 2019-03-19 by tatokis