ckb-next/ckb-next #389: Sniper Button / Custom Sensitivity in percentage depending on current mode.

Sur3, 2019-03-19:
I've the problem that the sniper button or custom button sensitivity is only settable in absolute values, but when playing in different dpi stage modes this absolute setting is quite useless. Therefore I request sniper button and custom dpi button settings in percentage depending on the current dpi mode.

ZirconiumX, 2019-03-22:
Unfortunately, this is not how the hardware works.

The sniper button is essentially its own DPI setting, so in order to achieve what you're asking, we would need to constantly change the hardware DPI profile for it depending on the profile you currently have enabled. That would place wear on the internal flash memory of the mouse, which we are reluctant to do.

Sur3, 2019-03-25:
Ok that's understandable, but when I start my computer the mouse seems to have different colors than when my pc is fully started, does that mean every time my computer starts the daemon writes the flash memory?

ZirconiumX, 2019-03-25:
No; having to write to flash every time you change colour would quickly
wear it away if you're animating at 60FPS.

Instead, RGB settings are held in mouse RAM, which is why they don't
persist across replugs.

When you press the "save to hardware" button, it does get written to flash,
but most devices only support a static colour. What you see when you plug
your mouse in is that flash RGB profile.